riscv-software-src/riscv-isa-sim #1335: Add CPI modeling for Spike (idea)

Wojciech-Bartczak, 2023-04-20:
Hi,

As part of my work with Spike, I have encountered a need to measure performance of certain programs using "Zicsr" extension.
As I have found out, Spike simulates CPI=1. This was mention is some other comments,
including #621 .

To model CPI != 1, I have developed code that bases on idea used to encode instructions.
I wonder is it worth to include it into Spike.
Looking at the comments, there is some drive from users regarding this feature.

The code is available here:
[cpi modeling branch](https://github.com/Wojciech-Bartczak/riscv-isa-sim/commit/664409179dc29dbc9dd8dc9f3c2c540f764edb55)

It can be easily integrated. Simple program to test it:

```
#include <cstdint>
#include <cstdio>

static const unsigned int MAX_REPS = 10;

struct perf_stamp
{
    void fetch() noexcept {
        asm __volatile__ ("rdcycle %[c]\n\t"
                          "rdinstret %[i]\n\t"
                          : [c]"=r"(cycle), [i]"=r"(instret));
    }

    uint64_t cycle;
    uint64_t instret;
};

#pragma GCC push_options
#pragma GCC optimize ("O0")
int main(int argc, char *argv[])
{
    struct perf_stamp begin {};
    struct perf_stamp end {};
    int64_t result = 0;
    int64_t a = 19;
    int64_t b = 5;

    begin.fetch();

    // Force a few multiplications
    for (unsigned int cnt = 0; cnt < MAX_REPS; cnt++) {
        result += a * b;
    }

    end.fetch();

    printf("cycles: %lu, instructions: %lu, result: %ld\n",
           end.cycle - begin.cycle,
           // -1 is due to extra instruction to get cycles
           end.instret - begin.instret - 1,
           result);

    return 0;
}
#pragma GCC pop_options
```

`riscv64-unknown-elf-g++ -Wall -O2 -g -o test_cpi cpi_modeling.cc`

Some technical details relevant for the patch:

I use header file _XXX_cycles.h_ for each instruction which is generated during compilation.
Code in header operates on mcycle performance counter 
`#define CONSUME_CYCLES(cycles) STATE.mcycle->bump((cycles))` .
Rest is done in compilation time. Depending on --enable-cpi-modeling switch,
CONSUME_CYCLES() uses value 1 or value provided by gen_instr_cycle.sh.

I understand that Spike doesn't emulate real HW. However, the change
doesn't aim to be 100% accurate. It still a simulation, just more precise simulation.
Cycles per instruction are just single number, they merely model reality.
For example branch instructions can't model taken or not taken branches.

Performance change caused by using this code should be minimal.
Mostly, compile time is impacted, due to call to script.
Runtime impact is related to multiple calls to wide_counter_csr_t::bump()
with constant value known at runtime. So, they should be optimized properly by compiler.
Previously wide_counter_csr_t::bump() has been called once per processor_t::step().

I appreciate all comments,
Wojciech Bartczak.

aswaterman, 2023-04-20:
It may be a minority opinion, but I’d prefer to keep Spike simple in this respect and avoid getting in the perf-modeling business altogether.

For the kinds of processors I’m interested in, even a very coarse perf model would require far more sophistication, and so would be better left to other simulators.

jerryz123, 2023-04-20:
I feel the best approach for this kind of modelling is a system that can post-process spike's commit log. Spike can be linked against as a library, and the commit trace can be determined using a C++ API. Decoupling the performance model from Spike would also make it easier to improve the fidelity of the model without stomping over Spike internals.

aswaterman, 2023-04-20:
Agreed.

Wojciech-Bartczak, 2023-04-21:
Thank you! I really appreciate your feedback.

closed 2023-04-21 by Wojciech-Bartczak

dogeDonge, 2025-01-25:
> I feel the best approach for this kind of modelling is a system that can post-process spike's commit log. Spike can be linked against as a library, and the commit trace can be determined using a C++ [...]

I want to do some analysis on the _logfile_ generated by Spike to get perf analysis, but I don't know how to implement this post-processing system. Could you elaborate on your ideas or methods for implementation? Or is there any reference case?