riscv-software-src/riscv-isa-sim #1336: Should execution loop return to interactive mode after firing a trigger?

YenHaoChen, 2023-04-21:
The execution loop returns to interactive mode after taking a trap, even without completing an instruction (https://github.com/riscv-software-src/riscv-isa-sim/commit/82b22f61c7e52eff8285d123a8cf9fd59df31425). This behavior allows the interactive mode to observe register differences before and after the trap.

Should the execution loop return to interactive mode after firing a trigger and allow the interactive mode to observe differences before and after the trigger action?

scottj97, 2023-04-24:
Doesn't firing a trigger count as taking a trap? Does it not already stop and return to interactive mode because of that?

YenHaoChen, 2023-04-25:
The execution loop in ```processor_t::step(size_t n)``` of ```execute.cc``` does not return to interactive mode after a breakpoint exception due to triggers. 

More specifically, calling ```take_trap(trap_t& t, reg_t epc)``` does not necessarily return to interactive mode. Instead, the statement ```n = instret``` in catching ```trap_t``` (and ```wait_for_interrupt_t```) satisfies the loop-breaking condition and returns to the interactive mode, which does not happen when catching ```triggers::matched_t``` (or ```trap_debug_mode```).