riscv-software-src/riscv-isa-sim #1340: Incorrect vectored jump for VSEI, VSI and VSTI

JJ-Gaisler, 2023-04-25:
Hi, I am currently writing a test suite for the RISC-V hypervisor extension and I just came across an issue when I ran one of the tests on spike. I have tried running the same scenario in QEMU, there getting different behavior.

The scenario is the following, I'm asserting all of the virtual supervisor IRQs using the hvip register. However, when taking the IRQ the vectored jump goes the VS cause in the vector table instead of the S cause. 

I am running the latest version of spike 19a7f343af6fa197f3234e0630d10bd1e8db9849

`spike -m0x80000000:0x10000000,0x20000000:0x1000,0x4000:0x1000,0xB000:0x1000 --isa rv64gch_zba_zbb_zbc_zbs_zfh_zbkb_zicsr_zifencei_sscofpmf_smepmp_zicbom -p1 main`
1. Enter M-mode:
2. Enable IRQ in all status registers.
3. Enable the VS irq in hie
4. Enable the irq in vsie
5. Delegate the IRQ using hideleg.
6. Setup vstvec in vectored mode.
7. Write hvip
8. Check the content of VSIP, for example bit sie (9) is set.
9. Enter VS mode
10. The IRQ occurs but the jump goes to vstvec+10*4 instead of vstec+9*4
11.  Repeat for all VS IRQ sources

Binary used to reproduce the VSTI IRQ, GitHub didn't like a file without an extension.
[main.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/11320774/main.txt)
```
Log output from the program: 

TRACE: [hart_ctrl.c:35] STARTING CPU 0
DEBUG: [trap_handling.c: 917] Setting up trap handlers
TRACE: [page_tables.c:341] Setting up page tables between 0x80000000 -> 0x87c00000
TRACE: [page_tables.c:355] Setting up page tables for UART 0x10000000
DEBUG: [hp_util_fn.c: 304] Setting satp: 0x8000100000081013 
DEBUG: [hp_util_fn.c: 307] Setting hgatp: 0x8000000000081010 
TRACE: [irq_trigger.c:179] Trying to cause VSTI
TRACE: [irq_trigger.c:182] Switching mode to VS
DEBUG: [trap_handling.c:  57] Environment call from M-mode exception: MEPC = 0x80004340, MTVAL = 0x0
DEBUG: [trap_handling.c: 731] Entered VSTVEC VSTI handler
sip
    ssip (1)   =  0
    vssip(2)   =  0
    msip (3)   =  0
    stip (5)   =  1
    vstip(6)   =  0
    mtip (7)   =  0
    seip (9)   =  0
    vseip(10)  =  0
    meip (11)  =  0
    sgeip(12)  =  0
ERROR: [trap_handling.c:733] VSTI in VS-mode should not be possible
TRACE: [trap_handling.c:28] Program done, exiting
```

aswaterman, 2023-04-25:
cc @avpatel

scottj97, 2023-04-25:
I need a little more info. What specific version (git hash) of Spike did you use? What was your command line?

When I run your `main.txt` on yesterday's master (7e82fd7d19) like so:
```
spike --isa=rv64gch -l --log-commits spike-pr-1340.elf 2>log
```

It fails like so:
```
*** FAILED *** (tohost = 11)
```

I found `tohost` at 0x8000a040 in the ELF's symbol table, and the fatal store comes in this sequence:
```
core   0: >>>>  m_save_context
core   0: 0x00000000800019d4 (0xf7810113) addi    sp, sp, -136                                                                                   core   0: 3 0x00000000800019d4 (0xf7810113) x2  0x000000008000a010                                                                               core   0: 0x00000000800019d8 (0x0000e006) c.sdsp  ra, 0(sp)
core   0: 3 0x00000000800019d8 (0xe006) mem 0x000000008000a010 0x0000000080000a86
core   0: 0x00000000800019da (0x0000e42a) c.sdsp  a0, 8(sp)
core   0: 3 0x00000000800019da (0xe42a) mem 0x000000008000a018 0x0000000000000002
core   0: 0x00000000800019dc (0x0000e82e) c.sdsp  a1, 16(sp)
core   0: 3 0x00000000800019dc (0xe82e) mem 0x000000008000a020 0x0000000000000000
core   0: 0x00000000800019de (0x0000ec32) c.sdsp  a2, 24(sp)
core   0: 3 0x00000000800019de (0xec32) mem 0x000000008000a028 0x0000000000000050
core   0: 0x00000000800019e0 (0x0000f036) c.sdsp  a3, 32(sp)
core   0: 3 0x00000000800019e0 (0xf036) mem 0x000000008000a030 0x0000000000000073
core   0: 0x00000000800019e2 (0x0000f43a) c.sdsp  a4, 40(sp)
core   0: 3 0x00000000800019e2 (0xf43a) mem 0x000000008000a038 0x00000000800092e0
core   0: 0x00000000800019e4 (0x0000f83e) c.sdsp  a5, 48(sp)
core   0: 3 0x00000000800019e4 (0xf83e) mem 0x000000008000a040 0x0000000000000017
core   0: 0x00000000800019e6 (0x0000fc42) c.sdsp  a6, 56(sp)
core   0: 3 0x00000000800019e6 (0xfc42) mem 0x000000008000a048 0x0000000000000000
```

Which does not look like an intentional test failure. (The 0x17 is shifted right by 1 to get the [decimal] 11 failure code.)

scottj97, 2023-04-25:
I believe you have identified a bug with the virtualized interrupts when `vstvec & 1`, i.e. Vectored.

This code decides the target PC (`vector`) and cause:
https://github.com/riscv-software-src/riscv-isa-sim/blob/87e0a1f86e16e4d3048e0afe7b2c51c821a108ed/riscv/processor.cc#L823-L825

The `cause` is adjusted by 1 for interrupts, but the target vector is not, and should be.

avpatel, 2023-04-26:
I agree with @scottj97 . This is a bug in `processor_t::take_trap()`.

JJ-Gaisler, 2023-04-26:
> I need a little more info. ~What specific version (git hash) of Spike did you use?~ What was your command line?
> 
> When I run your `main.txt` on the same commit you used ([19a7f34](https://github.com/riscv-software-src/riscv-isa-sim/commit/19a7f343af6fa197f3234e0630d10bd1e8db9849)) like so:
> 
> ```
> spike --isa=rv64gch -l --log-commits spike-pr-1340.elf 2>log
> ```
> 
> It fails like so:
> 
> ```
> *** FAILED *** (tohost = 11)
> ```
> 
> I found `tohost` at 0x8000a040 in the ELF's symbol table, and the fatal store comes in this sequence:
> 
> ```
> core   0: >>>>  m_save_context
> core   0: 0x00000000800019d4 (0xf7810113) addi    sp, sp, -136                                                                                   core   0: 3 0x00000000800019d4 (0xf7810113) x2  0x000000008000a010                                                                               core   0: 0x00000000800019d8 (0x0000e006) c.sdsp  ra, 0(sp)
> core   0: 3 0x00000000800019d8 (0xe006) mem 0x000000008000a010 0x0000000080000a86
> core   0: 0x00000000800019da (0x0000e42a) c.sdsp  a0, 8(sp)
> core   0: 3 0x00000000800019da (0xe42a) mem 0x000000008000a018 0x0000000000000002
> core   0: 0x00000000800019dc (0x0000e82e) c.sdsp  a1, 16(sp)
> core   0: 3 0x00000000800019dc (0xe82e) mem 0x000000008000a020 0x0000000000000000
> core   0: 0x00000000800019de (0x0000ec32) c.sdsp  a2, 24(sp)
> core   0: 3 0x00000000800019de (0xec32) mem 0x000000008000a028 0x0000000000000050
> core   0: 0x00000000800019e0 (0x0000f036) c.sdsp  a3, 32(sp)
> core   0: 3 0x00000000800019e0 (0xf036) mem 0x000000008000a030 0x0000000000000073
> core   0: 0x00000000800019e2 (0x0000f43a) c.sdsp  a4, 40(sp)
> core   0: 3 0x00000000800019e2 (0xf43a) mem 0x000000008000a038 0x00000000800092e0
> core   0: 0x00000000800019e4 (0x0000f83e) c.sdsp  a5, 48(sp)
> core   0: 3 0x00000000800019e4 (0xf83e) mem 0x000000008000a040 0x0000000000000017
> core   0: 0x00000000800019e6 (0x0000fc42) c.sdsp  a6, 56(sp)
> core   0: 3 0x00000000800019e6 (0xfc42) mem 0x000000008000a048 0x0000000000000000
> ```
> 
> Which does not look like an intentional test failure. (The 0x17 is shifted right by 1 to get the [decimal] 11 failure code.)
> 
> Edit: I also saw no console output before the one-line FAILURE message. Does this binary use the unofficial HTIF standard for console writes?

I am using this code to get console output [syscalls.c](https://github.com/riscv-software-src/riscv-tests/blob/master/benchmarks/common/syscalls.c )
As for tohost I am mainly using that to call tohost_exit and not to signal test success, but it looks like you must have gotten many recursive traps if the stack pointer has collided with tohost. It looks like you didn't run the binary with all of the ISA options I specified so I suppose you may have hit many illegal instructions.

Anyways, this may not be to relevant anymore as it seems like you agree that this is indeed a bug.

JJ-Gaisler, 2023-04-27:
> > I am using this code to get console output [syscalls.c](https://github.com/riscv-software-src/riscv-tests/blob/master/benchmarks/common/syscalls.c) As for tohost I am mainly using that to call tohost_exit and not to signal test success, but it looks like you must have gotten many recursive traps if the stack pointer has collided with tohost. It looks like you didn't run the binary with all of the ISA options I specified so I suppose you may have hit many illegal instructions.
> > Anyways, this may not be to relevant anymore as it seems like you agree that this is indeed a bug.
> 
> I would like to be able to recreate the failure that you saw, so I can be sure it's fixed. Can you tell me the command line that you used?

The command is in my original post:
``` 
spike -m0x80000000:0x10000000,0x20000000:0x1000,0x4000:0x1000,0xB000:0x1000 --isa rv64gch_zba_zbb_zbc_zbs_zfh_zbkb_zicsr_zifencei_sscofpmf_smepmp_zicbom -p1 main
```
Once patched I can come back and report if Spike passes all of my current tests.

scottj97, 2023-04-27:
> The command is in my original post:

Yeah sorry, don't know how I missed that. Fix submitted in PR #1345.

PR #1345 (2024-01-12): Fix vectored VS-level interrupts
Interrupts destined for VS-mode (i.e. VSEI, VSTI, and VSSI) are presented to the guest OS as supervisor-level interrupts (i.e. SEI, STI, and SSI), according to the hypervisor spec regarding `hideleg`:

> When a virtual supervisor external interrupt (code 10) is delegated to VS-mode, it is automatically translated by the machine into a supervisor external interrupt (code 9) for VS-mode, including the value written to vscause on an interrupt trap. Likewise, a virtual supervisor timer interrupt (6) is translated into a supervisor timer interrupt (5) for VS-mode, a [... cut]
commit 69ff70fc (2024-01-12): Fix vectored VS-level interrupts

They were going to the vector for cause 10 (VSEI) even though vscause
had been correctly translated to 9 (SEI).

Fixes #1340.
closed 2024-01-12 by aswaterman