riscv-software-src/riscv-isa-sim #1344: Spike not taking Illegal Instruction exception on vfredosum.vs(with vd=v0 and vm= 0)

Abdul-Vasan, 2023-04-27:
I am running spike with the following options: 
`--isa=rv64imafdcv_zba_zbb_zbc_zbs --priv=msu --misaligned --varch=vlen:256,elen:64`

I am running a test with vectors, from disassembly :
`829eda40:	0cae9057          	vfredosum.vs	v0,v10,v29,v0.t`

expecting spike to take illegal instruction exception but it is not:
```
core   0: 0x00000000829eda40 (0x0cae9057) vfredosum.vs v0, v10, v29, v0.t
core   0: 3 0x00000000829eda40 (0x0cae9057) e32 m2 l16 v0  0x00000000000000000000000000000000b4a89c9084786c6054483c307e6d661a c8_vstart 0x0000000000000000 c1_fflags 0x0000000000000011
```

According to RISC-V vector spec 5.3. Vector Masking:

> The destination vector register group for a masked vector instruction cannot overlap the source mask register (v0), unless the destination vector register is being written with a mask value (e.g., compares) or the scalar result of a reduction. These instruction encodings are reserved.

aswaterman, 2023-04-27:
Reductions are a special case.  As the top of Section 14 says, "The destination vector register can overlap the source operands, including the mask register."

Abdul-Vasan, 2023-04-27:
ok, Thank you for the clarity.

closed 2023-04-27 by Abdul-Vasan