riscv-software-src/riscv-isa-sim #1351: Zfa: fround vs froundnx

aamartin0000, 2023-05-08:
In Ch.25.3 of unpriv, FROUNDNX is defined as the same as FROUND, except that it sets NX if the input differs from the rounded result (and is not NaN).

In the Spike code, both of these instructions call softfloat/f*_roundToInt, while the "exact" operand is true for fround but false for froundnx.  Setting NX is done under some conditions only if the "exact" operand is true.  This appears to conflict with the definition in unpriv.

Am I understanding this incorrectly?  If not, is unpriv correct, or is Spike?

aswaterman, 2023-05-09:
The spec is unambiguously authoritative in this case.  I'm guessing there was a misunderstanding of the meaning of the `exact` argument to SoftFloat's `roundToInt` routines.  I agree it should be inverted at the call sites.

cc @ptomsich - can you add this to the list of Zfa issues that need to be resolved.

ptomsich, 2023-05-09:
I'll have to check the SAIL code as well, as I seem to have read "NX" as "_not_ exact".

aamartin0000, 2023-05-09:
Is this list available somewhere?  I am implementing Zfa in hardware, and need my counterpart to have the correct code in his (Spike-based) verification framework.

aswaterman, 2023-05-09:
@aamartin0000 the “list” is this GitHub issue tracker. Since the feature was only contributed a few weeks ago, you’ll only need to scan back through a handful of issues to find all of the relevant ones.

aswaterman, 2023-05-09:
@aamartin0000 I believe just #1312 and #1327, which @ptomsich will presumably handle sometime soon.

closed 2023-05-09 by aswaterman

aamartin0000, 2023-05-09:
Ok, thanks.  My verif counterpart told me of another discrepancy, with fleq/fltq.  I'll open a new issue for that.

aswaterman, 2023-05-09:
Cool.  It takes a village...

PR #1353 (2023-05-09): Zfa: fix NX handling for the fround/froundnx family
The initial implementation (together with the SAIL code and the tests) had gotten the NX variants backwards (as in 'an inexact result is ok' vs. 'set NX is inexact').  Update all 4 instructions.

Resolves #1351