riscv-software-src/riscv-isa-sim #1360: Wrong AMOXXX.D memory result for RV64.

matszpk, 2023-05-13:
I found unexpected and erroneous behaviour of AMO64 (atomic memory operations) if some conditions satisfied.
In details: these instructions store wrong result to memory if 31-bit of result is set. Then all higher 32-bits are bits are sets.
Example: `amoxor.d x5, x23, (x18)` store incorrect result to memory if: x5=0xda04520b33223a11, x23=0xc05a0b3fddac2601 then store value 0xffffffffee8e1c10 to memory address in x18. These bug is working for all AMOADD.D, AMOAND.D, AMOOR.D
and maybe of rest of AMO64 instructions.
I am using SPIKE 1.1.0 under 64-bit (AMD64) Linux (Arch Linux).

aswaterman, 2023-05-14:
I do not observe the same phenomenon when trying a similar case (with bit 31 set).  Please provide instructions to reproduce on HEAD of master.

aswaterman, 2023-05-15:
Thank you for reporting back!