riscv-software-src/riscv-isa-sim #1369: Shouldn't fcvtmod.w.d produce INT32_MIN/INT32_MAX when input outsize of range (but not Nan/Inf)?

mlawson-tt, 2023-05-26:
Currently the handler for `fcvtmod.w.d` in fcvtmod_w_d.h has explicit handling for overflow conditions (i.e. values that are not NaN/Inf with exponent values larger than 31), but in at least the case where the input double value has the largest valid exponent (1023 / exponent bits = 0x7fe), the resulting value is 0 (the working significand is set to 0 in the first `if` statement when checking the shift value`.  To match `fcvt.w.d`'s behavior in this case, shouldn't the resulting value be `INT32_MIN` if the sign bit is set and `INT32_MAX` otherwise?

aswaterman, 2023-05-27:
It's correct that the result be 0 in the case that the input has the largest valid exponent: emphasis on _unbounded_ in "bits 31:0 are taken from the rounded, unbounded two's complement result".

I haven't looked at the other cases, but I want to emphasize that this particular one is not a bug.

aswaterman, 2023-05-27:
(In general, the point of this instruction is that we _don't_ want to saturate to INT_MIN/INT_MAX.)

ptomsich, 2023-05-30:
Just for reference, the SAIL code for this instruction is at https://github.com/riscv/sail-riscv/blob/master/model/riscv_insts_zfa.sail#L762.
This shows the intermediate calculation as creating a 32.52 fixed-point intermediate and then checking for overflows (detecting NV) and whether the fractional bits are zero/non-zero (for NX).  While we still (note to self: this may be a future improvement of the SAIL code) don't generate an unbounded result (which would be easy enough by creating a 1023.1075 fixed-point intermediate before testing for overflows and fractionals) in the current SAIL implementation, the SAIL code already is a pretty good formal specification of the instruction.

mlawson-tt, 2023-05-30:
Yeah, thanks for pointing out the SAIL implementation.  Turns out that version was actually how I figured out what I was misunderstanding 😄 .

ptomsich, 2023-05-30:
We are actively working on including the SAIL code in the specification documents to provide additional clarity.

This will also explain why I worry about whether to express the "unbounded result" concept in SAIL by creating a very wide fixed-point intermediate... for an executable specification (such as the SAIL model) the clarity should be more important than any premature optimization of execution throughput.