riscv-software-src/riscv-isa-sim #1380: Question about setting spike

huikmn, 2023-06-14:
We are using the latest spike: Spike RISC-V ISA Simulator 1.1.1-dev as our reference mode. In our design, we choose all 1s as our agnostic policy for masked and tail elements. But the spike seems choose undisturbed as default, and we haven't find any settings to change this. So out result comparesion is always wrong. Is there any way to change the agnostic policy of spike to all 1s? Thank you!

aswaterman, 2023-06-14:
That feature is not currently provided. Note that providing it in a useful way for comparing against other microarchitectures is not straightforward. For example, some microarchitectures with register renaming will fill 1s up to the next VLEN boundary, but will leave any remaining vector registers within the vector register group undisturbed. Others might fill with 1s all the way to the end.

For the time being, I suggest modifying Spike and maintaining your own fork that has this alternate behavior.

huikmn, 2023-06-14:
> That feature is not currently provided. Note that providing it in a useful way for comparing against other microarchitectures is not straightforward. For example, some microarchitectures with [...]

I understand, Thank you

closed 2023-06-14 by huikmn