riscv-software-src/riscv-isa-sim #1395: vfmin/vfmax/vfredmin/vfredmax handle NaNs incorrectly

aswaterman, 2023-06-30:
Their implementations call the SoftFloat routines `min` and `max`, which don't do the same thing as the IEEE 754 `minimumNumber` and `maximumNumber` routines.

aswaterman, 2023-06-30:
Nevermind, I think the bug report was incorrect.  These SoftFloat routines do implement `minimumNumber` and `maximumNumber`.

closed 2023-06-30 by aswaterman

WojciechMula, 2023-06-30:
Soft float are correct, but implementation of vfredmin disabled NaN propagation:

```
commit 21dd7ed0aba0d58fb81f93b1063f54b42cc9b4f4
Author: Chih-Min Chao <chihmin.chao@sifive.com>
Date:   Mon May 11 23:35:38 2020 -0700

    rvv: handle inactive and NaN case for vfredsum

    Signed-off-by: Chih-Min Chao <chihmin.chao@sifive.com>

diff --git a/riscv/insns/vfredmin_vs.h b/riscv/insns/vfredmin_vs.h
index 37256cbc..e3cf1513 100644
--- a/riscv/insns/vfredmin_vs.h
+++ b/riscv/insns/vfredmin_vs.h
@@ -1,4 +1,5 @@
 // vfredmin vd, vs2, vs1
+bool is_propagate = false;
 VI_VFP_VV_LOOP_REDUCTION
 ({
   vd_0 = f16_min(vd_0, vs2);
```

aswaterman, 2023-06-30:
Can you provide a failing test case?  Simply calling e.g. `f16_min` repeatedly then assigning the result to the destination register should suffice--and that's what the code appears to do.

WojciechMula, 2023-06-30:
I think fpmin/fpmax is broken. The spec:

> For FMIN and FMAX, if at least one input is a signaling NaN, or if both inputs are quiet NaNs,
the result is the canonical NaN. If one operand is a quiet NaN and the other is not a NaN, the
result is the non-NaN operand.

Let's start with this:

```
$ cat snan.c
#include <stdio.h>

// this comes from riscv/riscv-isa-sim/softfloat/specialize.h
#define softfloat_isSigNaNF32UI( uiA ) ((((uiA) & 0x7FC00000) == 0x7F800000) && ((uiA) & 0x003FFFFF))

int main() {
    if (softfloat_isSigNaNF32UI(0x7fbfffff)) {
        puts("signaling NaN");
    }
}
$ gcc snan.c && ./a.out
signaling NaN
```

The simple asm code:

```
    vsetvli t0, zero, e32, m1

    li          a0, 0x7fbfffff      # signaling NaN
    li          a1, 0x3f000000      # 0.5

    vmv.s.x     v0, a0
    vmv.s.x     v1, a1

    vfmax.vv    v2, v0, v1
    vmv.x.s     t6, v2

    ld          zero, (zero) # to nicely print regs :)
```

Output is: `t6 000000003f000000`. This is wrong, SNaN should yield canoncial NaN (`0x7fc00000`)

I'll back to this issue the next week.

nick-knight, 2023-06-30:
@WojciechMula What version of the ISA spec are you looking at?

https://github.com/riscv/riscv-isa-manual/pull/75/files

https://github.com/riscv/riscv-isa-manual/releases

WojciechMula, 2023-07-03:
@nick-knight Oh, thanks a lot! I'm mostly focused on RVV extension. I grabbed RISC-V spec from some github repo, and didn't pay much attention if it's the really the freshest one.