riscv-software-src/riscv-isa-sim #1396: Debugging With Gdb: warning: rot13-64 has a LOAD segment with RWX permissions

WajahatRiaz, 2023-07-02:
I'm using the test program and files provided in the GitHub repository. I'm getting the following error: 

`/home/wajahatriaz/RISCV64/riscv64-unknown-elf-toolchain/riscv/bin/../lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: rot13-64 has a LOAD segment with RWX permissions
`
<br>Kindly look into the **README.md** file in the repository and please let me know if any changes are required. Furthermore, can you please point me to proper documentation as **riscv-isa-sim**.

WajahatRiaz, 2024-06-30:
Were you able to find a solution for this?

WajahatRiaz, 2024-06-30:
@aswaterman do you have any idea about this?

aswaterman, 2024-07-02:
Recent versions of binutils issue warnings if executable segments have write permissions.  This is to reduce the likelihood of security vulnerabilities in application code.

The message you quoted isn't an error; it is a warning.  In this context, you can safely ignore it.

closed 2024-10-01 by WajahatRiaz