riscv-software-src/riscv-isa-sim #1397: vs stage implicit memory access denied by PMP should cause a guest page fault or acess fault

GuoShibo-cn, 2023-07-04:
According to Hypervisor Extension, Version 1.0
[image]
Implicit memory access during first-stage(VS-stage) address translation, may arise guest page fault , what if the fault is caused by PMP deny? Should it rise an access fault or guest page fault. Recently Spike rise access fault.
  
[image]
[image]

scottj97, 2023-07-04:
PMP errors always throw access faults, never page faults. There's no such thing as a guest access fault.

GuoShibo-cn, 2023-07-10:
Get it. Thanks !

closed 2023-07-10 by GuoShibo-cn