riscv-software-src/riscv-isa-sim #1399: CM.JT raises and exception

ErdemTsyrenov, 2023-07-05:
# Issue
I run a program:
```
.global _start
_start:
  li      x1, 0x1
  la      x1, vector
  slli    x1, x1, 0x6
  csrw    jvt, zero
  csrw    jvt, x1
  cm.jt   0x0
  li      x1, 0x1

vector:
  li      x1, 0x0
```

When spike executes cm.jt instruction from Zcmt extension, I see instruction access fault exception:
```
core   0: 0x00010084 (0x01709073) csrw    jvt, ra
core   0: 0x00010088 (0x0000a002) cm.jt   0
core   0: exception trap_instruction_access_fault, epc 0x00010088
core   0:           tval 0x00402300
``` 

Could you please explain me a reason of that exception? Also I don't understand why `tval=0x00402300` which equals to JVT CSR value. According to specs it should be equal to fetch adress (i,e, `JVT[31:6]`) which is address of `vector` section in this case.

# Steps to reproduce
Compilation:
```
$ riscv32-unknown-elf-gcc -march=rv32ima_zca_zcmt -mabi=ilp32 -nostartfiles test.S -o test.elf
```  
Run:
```
$ spike --priv=m -l --isa=rv32ima_zca_zcmt -m0x10000:0x100000 test.elf
```

liweiwei90, 2023-07-05:
The base address for jump table is JVT[31:6] with 000000, not simple JVT[31:6].

ErdemTsyrenov, 2023-07-05:
Thanks for your reply! Now everything is clear to me.