riscv-software-src/riscv-isa-sim #1401: Memory address 0x0 is invalid

hyperion009, 2023-07-06:
Hi
My riscv core is used in a kind of  specific suitation. The program's address linked by ld is to be at 0x0.  However, I got following error messages:
Access exception occurred while loading payload hmrv_core_arithmetic_basic_test_0.o:
Memory address 0x0 is invalid

what else can I do to work around this issue?

thanks in advance.

liweiwei90, 2023-07-07:
The default DRAM_BASE is 0x80000000. You can specify memory layout by "-m" option.

scottj97, 2023-07-07:
I believe Spike has a debug ROM at 0. You might have to disable the external debug feature, which would require source code changes to Spike.

hyperion009, 2023-07-07:
@liweiwei90 thanks, something like -m 0x0:64, right?

hyperion009, 2023-07-07:
@liweiwei90 I appended -m 0:64 to spike and got a message:
Warning: the memory at [0x0, 0x3F] has been realigned
to the 4 KiB page size: [0x0, 0xFFF]
Access excpetion occurred while loading payload hmrv_core_arithmetic_basic_test_0.o:
Memory address 0x0 is invalid

hyperion009, 2023-07-07:
@scottj97 thanks scott， I am a newbie to spike. Could you provide more details on this, it's better to show me an example.

scottj97, 2023-07-19:
I'm afraid I don't know much about the debug hardware that's modeled in Spike, but it might be as simple as removing the debug device [here](https://github.com/riscv-software-src/riscv-isa-sim/blob/371353288e0c6a94713309c5400548f178648b7a/riscv/sim.cc#L72)

scottj97, 2023-07-19:
Or changing DEBUG_START to point to somewhere else.

timsifive, 2023-07-19:
> Or changing DEBUG_START to point to somewhere else.

A quick test shows that this breaks debugging.

scottj97, 2023-07-19:
> A quick test shows that this breaks debugging.

Not surprising. I'm assuming since OP has program code residing at address 0, he doesn't care about debugging.

hyperion009, 2023-07-24:
@scottj97 @timsifive thanks Scott&Tim. Finally we change our design to support the full address space.

xusq123, 2024-09-14:
this question has been solve?  how to deal with it