riscv-software-src/riscv-isa-sim #1404: mcountinhibit is not working

omerguzelelectronicguy, 2023-07-07:
I tried to change the `mcountinhibit` to test the `minstret` register. However, I couldn't change the `mcountinhibit` register. Does it a bug or an unnecessary feature for spike?

```
core   0: 0x00000000800000f4 (0x00004195) c.li    gp, 5
core   0: 3 0x00000000800000f4 (0x4195) x3  0x0000000000000005
(spike) 
core   0: 0x00000000800000f6 (0x32019073) csrw    mcountinhibit, gp
core   0: 3 0x00000000800000f6 (0x32019073)
(spike) reg 0 mcountinhibit
0x0000000000000000
(spike) ```

aswaterman, 2023-07-07:
It’s a WARL register that Spike chooses to hardwire to 0.

aswaterman, 2023-07-12:
It would be a little funky within Spike because Spike wires the instructions-retired counter and the cycle counter together. So we’d also need to wire those two mcountinhibit bits together, too.

We’d also need to handle the counter implementation carefully, since Spike uses it to decide when to switch which hart is being simulated.

Given these constraints, and given it’s an optional feature, I am inclined to keep it unimplemented. But that’s just my opinion.

scottj97, 2023-07-19:
`minstret` and `mcycle` have been separated since cd668d2f067d802879a7842bc3bed50fa61b2ead and I can't see where the main loop in `execute.cc` uses the CSR objects to count.

aswaterman, 2023-07-19:
Oops, you’re right; I was describing how it worked a long time ago. With that in mind, this feature should be fairly straightforward to add, and would be a useful contribution.

omerguzelelectronicguy, 2023-10-25:
I tired to handle it and couldn't understand how to add. I couldn't look for a long time. Now, I can try to add this functionality. [This](https://github.com/riscv-software-src/riscv-isa-sim/pull/1381) pr added some other inhibit and mask feature which looks like similar to my question. I can start by changing [these lines](https://github.com/rivosinc/riscv-isa-sim/blob/c927773dd1584d870dd60a1cf86c0a8f0d138dd4/riscv/processor.cc#L226C1-L227C6)
to add inhibit for the case when the `Smcntrpmf` is not enabled.

I am working in gitlab locally. I am not familiar with working the public projects like `riscv-isa-sim`. So, I may need some help and guadiance.