riscv-software-src/riscv-isa-sim #1408: Allow overriding `mem_t`

michalt, 2023-07-10:
Hi! We'd like to be able to inject custom implementation of `mem_t` (that contains just a single `char*` for its memory) when building `sim_t`. Currently this is not possible since the methods of `mem_t` are not `virtual`, and the implementation of `mem_t` is sparse (splits things into page-sized allocations and stores them in `std::map`). 

There are a few options how this could be supported:

1. Change `mem_t` to support both the current sparse version of memory as well as just a `char*` (through something like `std::variant` or a manual tag).
2. Change `mem_t` to be an abstract class, where its current implementation would move to its own class `sparse_mem_t` (similarly to what #1342 does).
3. Make the methods of `mem_t` virtual. Currently only `contents()` would really need to be virtual.

Let me know what you think about it and if you have any better ideas! Thanks in advance! 😃

michalt, 2023-07-12:
@aswaterman Any opinions/suggestions? I'd be happy to send a PR! 😃

aswaterman, 2023-07-12:
@jerryz123 you were looking into doing something along these lines recently, right? What did you end up doing?

jerryz123, 2023-07-12:
We didn't end up needing this feature.

I think approach 3) is the best, make the interfaces of `mem_t` virtual such that code which links to `libriscv` can provide their own implementation to pass to `sim_t`.

aswaterman, 2023-07-12:
If that does actually suffice for @michalt’s use case, I’m fine with it. @michalt care to make a PR? If you prefer, we can hold off on merging it until you’re sure it’s sufficient.

michalt, 2023-07-13:
Sounds good, I'll test things on my end and send you a PR. Thanks!

closed 2023-07-20 by jerryz123

michalt, 2023-07-21:
Thanks for help everyone! 😄

PR #1415 (2023-07-20): Make methods of `mem_t` virtual to allow overriding
This allows the users to provide their own custom implementations of `mem_t` when constructing `sim_t`.

Fixes #1408.
commit cb013516 (2023-07-20): Introduce `abstract_mem_t` to allow custom implementations

This change allows to create custom implementations of `abstract_mem_t`
and inject them when constructing `sim_t`. The current `mem_t`
implementation remains unchanged.

Fixes #1408.