riscv-software-src/riscv-isa-sim #1412: Cycle Accurate Count for Programs involving Vector Processor Execution

dkmtnaura, 2023-07-12:
I understand that SPIKE is a functional simulator, hence the cycle accuracy is not guaranteed.

The simulator is functionally simulating the RISCV + Vector Processor execution, which means that the dependencies between instructions are correctly simulated. Actual cycles for vector part of processing depends on the hardware, so it is not being simulated.
Is there still a way to at least manually calculate the cycles using the cycles information obtained in the SPIKE simulator? If yes, how to do it?
Thank you

jerryz123, 2023-07-12:
It sounds like you want a performance model of some vector micro-architecture you have in mind. I think the answer to your question depends completely on the micro-architecture you want to model, and the level of fidelity you intend the model to capture. For any non-trivial implementation, I don't think its possible to calculate a useful performance estimate using only the existing CPI=1 cycles counter in spike.

dkmtnaura, 2023-07-13:
Thank you for feedback. 
Is it possible to get the assembly list file on SPIKE as a result of creating the executable?
While doing single stepping we can see the assembly instructions on the screen.
Looking at Assembly LST file and also having an understanding of micro-architecture, would it be possible to manually calculate the numbers by replacing cycle number (= 1 per instruction) to suitable number, having an understanding of what instructions get execute in parallel and what is its latency. 
I am just trying to see if there is a way to assess the numbers looking at SPIKE numbers.

aswaterman, 2023-07-13:
The `-l` command-line argument will emit that information. And yes, for some microarchitectures, that log might contain enough information to build a useful performance model. (Note it doesn’t include the data memory addresses for loads and stores, so it doesn’t suffice for modeling caches.)

aswaterman, 2023-07-17:
Yeah.

dogeDonge, 2025-01-25:
> I understand that SPIKE is a functional simulator, hence the cycle accuracy is not guaranteed.
>
> The simulator is functionally simulating the RISCV + Vector Processor execution, which means that the [...]

I'd like to do something similar with my own instruction set, but I don't know how to implement it. Can you give more details on how you ended up implementing performance modeling?