riscv-software-src/riscv-isa-sim #1414: Inconsistent rtldump format for same instruction.

eflaner, 2023-07-13:
While trying with `KHM8` instruction of `p` extension, the rtldump produces a weird formatted output for different set of input registers

Below snippet 
>  lui     x13,0x38
> lui     x23,0x8
> .4byte  0x8f768c77
> sw      x24,0(x31)
> sw      x25,4(x31)

generates rtldump as follows:

> core   0: 3 0x80000010 (0x8f768c77) c9_vxsat 0x00000001 x24 0x00007f00 c768_mstatus 0x00000000
> core   0: 3 0x80000014 (0x018fa023) mem 0x80002110 0x00007f00
> core   0: 3 0x80000018 (0x019fa223) mem 0x80002114 0x00000000

whereas, the below snippet

> lui     x21,0x3800
> lui     x29,0x3800
> .4byte  0x8fda8b77
> sw      x22,8(x31)
> sw      x23,12(x31)

generates rtldump as follows:

> core   0: 3 0x80000024 (0x8fda8b77) x22 0x007f0000 c9_vxsat 0x00000001 c768_mstatus 0x00000000
> core   0: 3 0x80000028 (0x016fa423) mem 0x80002118 0x007f0000
> core   0: 3 0x8000002c (0x017fa623) mem 0x8000211c 0x00008000

Observe the `c9_vxsat` in both rtldump's which is weirdly formatted. 

⚠️ Since ELF's aren't allowed to be attached, I've attached the elf as zip, rename the [dut.zip](https://github.com/riscv-software-src/riscv-isa-sim/files/12038157/dut.zip) as dut.elf 
and execute by calling `spike --log spike.dump --log-commits --isa=rv32icp +signature=spike.sig --signature-granularity=4 /tmp/dut.elf` will generate spike.dump which can be used to observe the full rtldump.

aswaterman, 2023-07-13:
What's `rtldump`?  The token `rtl` doesn't show up anywhere in the source code.

eflaner, 2023-07-13:
By rtldump -- i meant, the `commit log`  which is enabled during spike build using `--enable-commitlog` 
that generates (when using --log-commits) the log for the retired/committed instruction

aswaterman, 2023-07-13:
Gotcha.  Is the issue that we are using an unordered collection to record register writes, such that traversal order is unspecified?  If so, this might just be a matter of deleting an `unordered_` somewhere.

eflaner, 2023-07-14:
I've replaced [this unordered_map](https://github.com/riscv-software-src/riscv-isa-sim/blob/e85d2923a5e4fdd7d61cd0cc9f0685c97f451769/riscv/processor.h#L69) to map and rebuilt. 

Now I'm getting the commit log as 

> `core 0: 3 0x80000010 (0x8f768c77) c9_vxsat 0x00000001 x24 0x00007f00 c768_mstatus 0x00000000`

is there a way to swap the reg write and csr write -- expecting something like below:

> `core 0: 3 0x80000010 (0x8f768c77) x24 0x00007f00 c9_vxsat 0x00000001  c768_mstatus 0x00000000`

chihminchao, 2023-07-14:
@eflaner you could adjust the print helper function at https://github.com/riscv-software-src/riscv-isa-sim/blob/e85d2923a5e4fdd7d61cd0cc9f0685c97f451769/riscv/execute.cc#L63 to decide your order.
The other way is to provide customized comparison function to map container.

scottj97, 2023-07-19:
I don't believe the register writes shown in the commit log have any specified order. They all happen simultaneously, from the programmer's viewpoint.

If you're parsing the commit log, you should handle any needed sorting at that point.

closed 2023-07-20 by eflaner