riscv-software-src/riscv-isa-sim #1418: xtval update values on cross-boundary instruction exceptions

poemonsense, 2023-07-18:
Spike just makes me confused when co-simulating it with rocket. I assume rocket should be correct while Spike not.

Here is the logs:

```
// The last instruction is a ret, and the CPU jumps to 0x1fffe with virtual memory enabled.
// These commit logs are from Rocket
[19] commit pc 000000000001d8fe inst 00008082 wen 1 dst 00 data 0000000000000000 ret
[20] exception pc 000000000001fffe inst 0000ffff cause 0000000000000001 unknown <--

// Here comes the Spike logs.
// This is the last ret instruction
[Spike] pc = 0x1d8fe inst 0x8082

// Starting fetching the next instruction in refill_cache
[Spike] refill_icache addr: 1fffe 

// Fetch the first two bytes
[Spike] mem_ptw   addr: 0x8280b000, data: 0x0000000020a02401, len: 8
[Spike] ptw: level 2, vaddr 0x1fffe, pg_base 0x8280b000, p_pte 0x8280b000, pte.val 0x20a02401
[Spike] mem_ptw   addr: 0x82809000, data: 0x0000000020a03401, len: 8
[Spike] ptw: level 1, vaddr 0x1fffe, pg_base 0x82809000, p_pte 0x82809000, pte.val 0x20a03401
[Spike] mem_ptw   addr: 0x8280d0f8, data: 0x0000000020eacc5b, len: 8
[Spike] ptw: level 0, vaddr 0x1fffe, pg_base 0x8280d000, p_pte 0x8280d0f8, pte.val 0x20eacc5b

// after fetching the first two bytes, it returns 0xffff
// insn_length gives 8 so it tries to fetch the following 6 bytes in the orders of 6, 4, 2
[Spike] refill_icache addr: 1fffe length: 8, insn ffff  

// Now it starts fetching the following bytes (in the next-page) at base+6
[Spike] mem_ptw   addr: 0x8280b000, data: 0x0000000020a02401, len: 8
[Spike] ptw: level 2, vaddr 0x20004, pg_base 0x8280b000, p_pte 0x8280b000, pte.val 0x20a02401
[Spike] mem_ptw   addr: 0x82809000, data: 0x0000000020a03401, len: 8
[Spike] ptw: level 1, vaddr 0x20004, pg_base 0x82809000, p_pte 0x82809000, pte.val 0x20a03401
[Spike] mem_ptw   addr: 0x8280d100, data: 0x003fbfb4ca78ffff, len: 8
[Spike] ptw: level 0, vaddr 0x20004, pg_base 0x8280d000, p_pte 0x8280d100, pte.val 0x3fbfb4ca78ffff

// Now the translated address is out-of-range, and both Spike and rocket raise access-fault
```

However, their mtval are different. Here is the report:
```
pc: 0x00000000800a0004 mstatus: 0x8000000a000060a2 mcause: 0x0000000000000001 mepc: 0x000000000001fffe
                       sstatus: 0x8000000200006022 scause: 0x000000000000000f sepc: 0x00000000000c1126
satp: 0x800000000008280b
mip: 0x0000000000000000 mie: 0x000000000000002a mscratch: 0x00000000800abdc0 sscratch: 0x00000000800abdc0
mideleg: 0x0000000000000222 medeleg: 0x000000000000b109
mtval: 0x0000000000020004 stval: 0x00000000409a1078 mtvec: 0x00000000800a0004 stvec: 0xffffffe000200aa0
privilege mode:3
priviledgeMode: 3
  mtval different at pc = 0x000001d8fe, right= 0x0000000000020004, wrong = 0x0000000000020000
Core 0: ABORT at pc = 0x20002
```

I believe rocket is correct (`wrong = 0x0000000000020000`) and the RISC-V manuals require that xtval is set to the first address that causes exceptions. However, when I look the the ISA, it only says `mtval will contain the virtual address of the portion of the instruction that caused the fault`. Then I become confused about this case.

Let me summarize the case. (1) It should be a cross-boundary access-fault (virtual memory is either enabled or disabled? It seems both would cause the mismatch?), and (2) the accessed bits of the first two bytes should be decoded into length of 6 or 8 under the following macro. With these two conditions, Spike would access the highest bits first and then the lower ones, and then sets the xtval to the highest address at `base + 6` or `base + 8` given length of 6 or 8.

```
#define insn_length(x) \
  (((x) & 0x03) < 0x03 ? 2 : \
   ((x) & 0x1f) < 0x1f ? 4 : \
   ((x) & 0x3f) < 0x3f ? 6 : \
   8)

// In mmu.h: inline icache_entry_t* refill_icache(reg_t addr, icache_entry_t* entry)
// It always fetches the highest bytes first
if (likely(length == 4)) {
  insn |= (insn_bits_t)from_le(*(const uint16_t*)translate_insn_addr_to_host(addr + 2)) << 16;
} else if (length == 2) {
  // entire instruction already fetched
} else if (length == 6) {
  insn |= (insn_bits_t)from_le(*(const uint16_t*)translate_insn_addr_to_host(addr + 4)) << 32;
  insn |= (insn_bits_t)from_le(*(const uint16_t*)translate_insn_addr_to_host(addr + 2)) << 16;
} else {
  static_assert(sizeof(insn_bits_t) == 8, "insn_bits_t must be uint64_t");
  insn |= (insn_bits_t)from_le(*(const uint16_t*)translate_insn_addr_to_host(addr + 6)) << 48;
  insn |= (insn_bits_t)from_le(*(const uint16_t*)translate_insn_addr_to_host(addr + 4)) << 32;
  insn |= (insn_bits_t)from_le(*(const uint16_t*)translate_insn_addr_to_host(addr + 2)) << 16;
}
```

So should it be an implementation-dependent feature of the RISC-V ISA or an issue/bug of the Spike? Could you please help make a clarification? @aswaterman

poemonsense, 2023-07-18:
After looking at the git history, I believe this issue is introduced at the first time multiple-byte fetch is implemented at https://github.com/riscv-software-src/riscv-isa-sim/commit/d643e43dca834fcb2e40504f65d5043d9f2b018b.

Still cannot tell whether it is a feature or bug yet. Much appreciate it if you could help clarify this.

poemonsense, 2023-07-18:
Besides, would cross-boundary instruction page-faults have the same issue?

poemonsense, 2023-07-18:
Also, if 48-bit, 64-bit, and wider instructions are not supported, which I believe matches the case for the current Spike, should it try to fetch more than four bytes?

For example, if the processor supports only 16bit or 32bit instructions, I believe it should never try to fetch more than four bytes on instruction accesses. So it should never raise access-faults on base+4 and the following bytes.

aswaterman, 2023-07-18:
Two comments:
- Spike is probably technically wrong to be fetching in reverse order, which causes mtval to be set to the greatest bad address, rather than the smallest.  I assume the correct fix is to reverse the statement order in the 6-byte and 8-byte fetch cases.
- Spike does support 48- and 64-bit instructions as custom extensions.  It's true that only 32-bit standard instructions are currently defined.

Anyway, it's a minor issue, since the OS will still do the right thing (mtval just needs to point to some address on the faulting page for the page-fault handler to do the right thing).

poemonsense, 2023-07-18:
Sure, and I agree with you that this generally would not bring issues to the software.

It would be better that the ISA could make it clear whether it is acceptable.

aswaterman, 2023-07-18:
Closing as completed.

closed 2023-07-18 by aswaterman