riscv-software-src/riscv-isa-sim #1420: MMIO plugin - Example needed

xzj478, 2023-07-18:
I am facing issues when trying to use the MMIO example, the code deviated quite a bit from the initial version also published at [0]. I tried multiple approaches to use it, but could not get it to work. Could you by any chance provide a minumum working example as provided in the linked repository?

Thank you!

[0] https://github.com/vexingcodes/spike-plugin

jerryz123, 2023-07-18:
I have an example implementing the SiFive UART. I'll try to get around to pushing this into another repo today.

xzj478, 2023-07-21:
Thanks you for your reply, I compiled it and tried to test it with the test driver provided in this repo: https://github.com/vexingcodes/spike-plugin (not sure if this would work), but it seems failed, maybe because the command line was wrong or I just need to write a new test file. Sorry, I'm new to RISCV(spike) simulator and I'm still learning, so I'm kindly asking: could you provide an assembly test file for the example you posted? Also, a correct command line to test the output with spike(like the one they did in spike-plugin repo) would be appreciated. Thank you so much.