riscv-software-src/riscv-isa-sim #1421: Some bitmanip instruction function in spike

alextxia13, 2023-07-20:
For RISC-V Bit-Manipulation ISA-extensions spec, rori, binvi, bclri bseti and bexti instructions have different encoding in RV64 and RV32. In the spec, it said for RV32, the encodings corresponding to shamt[5] =1 are reserved. How does spike work in this situation?
I see in rori.h, it requires (SHAMT > xlen) which means it will cause exception if shamt[5] =1. But in binvi, bclri bseti and bexti, it will not raise any exceptions. It seems that shamt[5] =1 is ignored. It is a little confused when the spike executes the instruction.

aswaterman, 2023-07-21:
This might be an omission.  It's technically not a bug, because raising illegal-instruction exceptions on reserved encodings is only recommended, not required.  But I don't think that was our intent in this case.  If you'd like to make a PR that makes these instructions match `rori`, we'd accept it.

kassasAndes, 2025-01-27:
https://github.com/riscv-software-src/riscv-isa-sim/pull/1907

The above PR has a fix for the shamt condition for zbs immediate instructions.

aswaterman, 2025-01-27:
Resolved by #1907

closed 2025-01-27 by aswaterman