riscv-software-src/riscv-isa-sim #1423: make:"../fesvr/option_parser.cc:" error

duranteen, 2023-07-21:
```
$ apt-get install device-tree-compiler
$ mkdir build
$ cd build
$ ../configure --prefix=$RISCV
$ make
$ [sudo] make install
```
When I execute `make` in the build directory, I get the following error:
[image]
I found spike_dasm_option_parser.cc, which contains:
`../fesvr/option_parser.cc`
I tried to change it to:
`#include "../fesvr/option_parser.cc"`
Then execute `make,` the error does not appear. Is this the correct solution? Or did I do it wrong in the first place.

scottj97, 2023-07-24:
`spike_dasm/spike_dasm_option_parser.cc` is supposed to be a symlink pointing to `../fesvr/option_parser.cc`.

Something's gone awry with your `git clone`.

aswaterman, 2023-07-24:
Could be a file system that doesn't support symbolic links (e.g. FAT32 if that's still in use).