riscv-software-src/riscv-isa-sim #1424: Issue during simulating elf file

SyedHassanUlHaq, 2023-07-24:
I want to simulate vector assembly on Spike, but Spike is unable to load the ELF file. I suspect the issue is with my ELF file being relocatable. Here are my toolchain and Spike configurations:

Spike configuration:

```plaintext
../configure --prefix=$HOME/riscv-isa-sim/build --with-varch=vlen:128,elen=32 --with-isa=rv32gcv --with-target=riscv32-unknown-elf
```

Toolchain Configuration:

```plaintext
../configure --prefix=$HOME/riscv-gnu-toolchain/build --with-arch=rv32gcv --with-abi=ilp32 --enable-multilib
```

To generate the ELF file, I used the command:

```plaintext
riscv32-unknown-elf-gcc -march=rv32gcv -mabi=ilp32 -c -o /home/hassan/add2vector.o /home/hassan/add2vector.c
```

To generate the log file using the ELF file, I used the command:

```plaintext
spike --varch=vlen:128,elen:32 --isa=rv32gcv add2vector.o logfile.log
```

The error that Spike shows is:

```
spike --varch=vlen:128,elen:32 --isa=rv32gcv add2vector.o logfile.log
spike: ../fesvr/elfloader.cc:43: std::map<std::__cxx11::basic_string<char>, long unsigned int> load_elf(const char*, memif_t*, reg_t*, unsigned int): Assertion `IS_ELF_EXEC(*eh64)' failed.
```

The terminal shows my add2vector.o file as:

```
add2vector.o: ELF 32-bit LSB relocatable, UCB RISC-V, RVC, soft-float ABI, version 1 (SYSV), not stripped
```

Any Suggestions ??

samadpls, 2023-07-24:
Try running the command without the '.o' extension, so your command will be:

```
riscv32-unknown-elf-gcc -march=rv32gcv -mabi=ilp32 -c -o /home/hassan/add2vector /home/hassan/add2vector.c
```

scottj97, 2023-07-25:
Also remove `-c` from the compile command line, which says "only compile; don't link". Spike needs a linked ELF, not an intermediate object file (`*.o`).

closed 2023-07-25 by scottj97