riscv-software-src/riscv-isa-sim #1425: Error Generating ELF file

SyedHassanUlHaq, 2023-07-24:
I've been trying to generate a elf file using riscv-gnu-toolchain rvv branch from the following vector code in C
```plaintext
#include <stddef.h>
#include <riscv_vector.h>

void saxpy(size_t n, const float a, const float *x, float *y) {
    size_t vl;
    vfloat32m8_t vx, vy;
    for (; n > 0; n -= vl) {
        vl = __riscv_vsetvl_e32m8(n);
        vx = vle32_v_f32m8(x, vl);
        vy = vle32_v_f32m8(y, vl);
        vy = vfmsac_vf_f32m8(vy, a, vx, vl);
        vse32_v_f32m8(y, vy, vl);
        x += vl;
        y += vl;
    }
}
```
Here are the sequential flow of commands that I used to configure and build the toolchain
```plaintext
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-next
```
```plaintext
cd riscv-gnu-toolchain
```
```plaintext
cd riscv-gcc 
```
```plaintext
git checkout -b riscv-gcc-rvv-next
```
```plaintext
git fetch
```
```plaintext
cd ..
```
```plaintext
mkdir build && cd build
```
```plaintext
../configure --prefix=$HOME/riscv-gnu-toolchain/build -with-arch=rv32gcv -with-abi=ilp32 -enable-multilib
```
```plaintext
sudo make
```

But when I try to generate the elf file using the command:
```plaintext
riscv32-unknown-elf-gcc -march=rv32i -mabi=ilp32 -o $HOME/add2vector $HOME/add2vector.c
```
It gives me an ERROR:
```plaintext
In file included from /home/hassan/add2vector.c:2:
/home/hassan/riscv-gnu-toolchain/build/lib/gcc/riscv32-unknown-elf/12.0.1/include/riscv_vector.h:29:2: error: #error "Vector intrinsics require the vector extension."
   29 | #error "Vector intrinsics require the vector extension."
      |  ^~~~~
/home/hassan/add2vector.c: In function 'saxpy':
/home/hassan/add2vector.c:6:5: error: unknown type name 'vfloat32m8_t'
    6 |     vfloat32m8_t vx, vy;
      |     ^~~~~~~~~~~~
/home/hassan/add2vector.c:8:14: warning: implicit declaration of function '__riscv_vsetvl_e32m8' [-Wimplicit-function-declaration]
    8 |         vl = __riscv_vsetvl_e32m8(n);
      |              ^~~~~~~~~~~~~~~~~~~~
/home/hassan/add2vector.c:9:14: warning: implicit declaration of function 'vle32_v_f32m8' [-Wimplicit-function-declaration]
    9 |         vx = vle32_v_f32m8(x, vl);
      |              ^~~~~~~~~~~~~
/home/hassan/add2vector.c:11:14: warning: implicit declaration of function 'vfmsac_vf_f32m8' [-Wimplicit-function-declaration]
   11 |         vy = vfmsac_vf_f32m8(vy, a, vx, vl);
      |              ^~~~~~~~~~~~~~~
/home/hassan/add2vector.c:12:9: warning: implicit declaration of function 'vse32_v_f32m8' [-Wimplicit-function-declaration]
   12 |         vse32_v_f32m8(y, vy, vl);
      |         ^~~~~~~~~~~~~
```

**Please tell me if there are any errors in my C code, configuration or anything else as I want to make generate a log file using spike**

aswaterman, 2023-07-24:
We can't provide support on debugging code in this forum.  However, I will note that your `-march` string is `rv32i` -- i.e., it excludes the V extension.

closed 2023-07-24 by aswaterman

SyedHassanUlHaq, 2023-07-25:
> I've been trying to generate a elf file using riscv-gnu-toolchain rvv branch from the following vector code in C
> 
> ```
> #include <stdio.h>  // For printf
> #include <stddef.h>
> #include <riscv_vector.h>
> 
> // Function to perform SAXPY operation using RISC-V vector instructions
> void saxpy(size_t n, const float a, const float *x, float *y) {
>     size_t vl;
>     vfloat32m8_t vx, vy;
>     for (; n > 0; n -= vl) {
>         vl = vsetvl_e32m8(n);
>         vx = vle32_v_f32m8(x, vl);
>         vy = vle32_v_f32m8(y, vl);
>         vy = vfmsac_vf_f32m8(vy, a, vx, vl);
>         vse32_v_f32m8(y, vy, vl);
>         x += vl;
>         y += vl;
>     }
> }
> 
> int main() {
>     size_t n = 10; // Number of elements in the vectors
>     const float a = 2.0; // Scalar constant
>     float x[10] = {1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0, 10.0}; // Input vector x
>     float y[10] = {0.0}; // Output vector y, initialized to 0
> 
>     saxpy(n, a, x, y);
> 
>     // Print the results of the SAXPY operation
>     printf("Resultant vector y: ");
>     for (size_t i = 0; i < n; i++) {
>         printf("%f ", y[i]);
>     }
>     printf("\n");
> 
>     return 0;
> }
> ```
> 
> Here are the sequential flow of commands that I used to configure and build spike
> 
> ```
> git clone https://github.com/riscv/riscv-isa-sim.git
> ```
> 
> ```
> cd riscv-isa-sim
> ```
> 
> ```
> mkdir build && cd build
> ```
> 
> ```
> ../configure --prefix=/home/utm/opt/riscvv --with-varch=vlen:128,elen=32 --with-isa=rv32gcv --with-target=riscv32-unknown-elf
> ```
> 
> ```
> make
> ```
> 
> ```
> make install
> ```
> 
> But when I try to generate the log file using the command:
> 
> ```
> spike --varch=vlen:128,elen:32 --isa=rv32gcv add2vector logfile.log
> ```
> 
> It gives me an ERROR:
> 
> ```
> Access exception occurred while loading payload add2vector:
> Memory address 0x10000 is invalid
> ```
> 
> **Please tell me if there are any errors in my C code, configuration or anything else as I want to make generate a log file using spike**