riscv-software-src/riscv-isa-sim #1426: Spike returns memory address invalid error

SyedHassanUlHaq, 2023-07-25:
> I've been trying to generate a elf file using riscv-gnu-toolchain rvv branch from the following vector code in C
> 
> ```
> #include <stdio.h>  // For printf
> #include <stddef.h>
> #include <riscv_vector.h>
> 
> // Function to perform SAXPY operation using RISC-V vector instructions
> void saxpy(size_t n, const float a, const float *x, float *y) {
>     size_t vl;
>     vfloat32m8_t vx, vy;
>     for (; n > 0; n -= vl) {
>         vl = vsetvl_e32m8(n);
>         vx = vle32_v_f32m8(x, vl);
>         vy = vle32_v_f32m8(y, vl);
>         vy = vfmsac_vf_f32m8(vy, a, vx, vl);
>         vse32_v_f32m8(y, vy, vl);
>         x += vl;
>         y += vl;
>     }
> }
> 
> int main() {
>     size_t n = 10; // Number of elements in the vectors
>     const float a = 2.0; // Scalar constant
>     float x[10] = {1.0, 2.0, 3.0, 4.0, 5.0, 6.0, 7.0, 8.0, 9.0, 10.0}; // Input vector x
>     float y[10] = {0.0}; // Output vector y, initialized to 0
> 
>     saxpy(n, a, x, y);
> 
>     // Print the results of the SAXPY operation
>     printf("Resultant vector y: ");
>     for (size_t i = 0; i < n; i++) {
>         printf("%f ", y[i]);
>     }
>     printf("\n");
> 
>     return 0;
> }
> ```
> 
> Here are the sequential flow of commands that I used to configure and build spike
> 
> ```
> git clone https://github.com/riscv/riscv-isa-sim.git
> ```
> 
> ```
> cd riscv-isa-sim
> ```
> 
> ```
> mkdir build && cd build
> ```
> 
> ```
> ../configure --prefix=/home/utm/opt/riscvv --with-varch=vlen:128,elen=32 --with-isa=rv32gcv --with-target=riscv32-unknown-elf
> ```
> 
> ```
> make
> ```
> 
> ```
> make install
> ```
> 
> But when I try to generate the log file using the command:
> 
> ```
> spike --varch=vlen:128,elen:32 --isa=rv32gcv add2vector logfile.log
> ```
> 
> It gives me an ERROR:
> 
> ```
> Access exception occurred while loading payload add2vector:
> Memory address 0x10000 is invalid
> ```
> 
> **Please tell me if there are any errors in my C code, configuration or anything else as I want to make generate a log file using spike**

scottj97, 2023-07-25:
Use the `-m` flag to teach Spike about your memory map.

```
  -m<a:m,b:n,...>       Provide memory regions of size m and n bytes                                                                         
                          at base addresses a and b (with 4 KiB alignment)                                                                   
```

SyedHassanUlHaq, 2023-07-25:
I've tried it but still it shows the same error, can you please edit my command that generates the log file maybe I am making a syntax error

scottj97, 2023-07-25:
You need to demonstrate that you've made some effort to diagnose and solve the problem, and stop expecting other people to do all your work for you.

What exact command line(s) did you try? What exact error message(s) did you receive?

SyedHassanUlHaq, 2023-07-25:
yeah I was making a mistake in the syntax of flag(was not placing it correctly in the command), just examined my command again and fixed it, it works now.
Thanks @scottj97

closed 2023-07-25 by SyedHassanUlHaq

caicaixuehua, 2023-09-24:
> yeah I was making a mistake in the syntax of flag(was not placing it correctly in the command), just examined my command again and fixed it, it works now. Thanks @scottj97

Hi I access the same issues, how do you solve it ?

SyedHassanUlHaq, 2023-09-24:
use the -m flag in command

vivekvpandya, 2024-01-16:
@SyedHassanUlHaq running into similar issue, here is my command
```
./spike --isa=RV32IM -m0:16777216,16777216:251658240,268435456:268435456,536870912:536870912,1073741824:268435456,1342177280:2684354560  /.../debug/fibonacci
Access exception occurred while loading payload /.../debug/fibonacci:
Memory address 0x10000000 is invalid
```
is there anything wrong in way I have used -m flag?

vivekvpandya, 2024-01-16:
Based on https://github.com/riscv-software-src/riscv-isa-sim/issues/1179#issuecomment-1352402114 it seems I need to change my memory layout, perhaps?

omerguzelelectronicguy, 2024-01-16:
@vivekvpandya So surprisingly, we are trying to do the same thing in the same hours. :)

SyedHassanUlHaq, 2024-01-16:
The memory starting address and range in the -m flag should be according to as described in the linker you are using to compile your program

SyedHassanUlHaq, 2024-01-16:
The format of the -m flag in Spike is <start_address>:<size>. Let's revisit the memory regions you specified:
```
- `0:16777216`: This defines a memory region starting from address 0 with a size of 16 MB.
- `16777216:251658240`: The second memory region starts from address 16777216 with a size of 240 MB.
- `268435456:268435456`: The third memory region starts from address 268435456 with a size of 256 MB.
- `536870912:536870912`: The fourth memory region starts from address 536870912 with a size of 512 MB.
- `1073741824:268435456`: The fifth memory region starts from address 1073741824 with a size of 256 MB.
- `1342177280:2684354560`: The sixth memory region starts from address 1342177280 with a size of 2.5 GB.
```
If you've followed this format and are still encountering the "Memory address 0x10000000 is invalid" error, there might be something missing or incorrect in your compilation process. Make sure that the memory regions specified in the -m flag align with the memory layout defined in your linker script. Additionally, check for any issues in your compilation and linking steps that might be affecting the generation of the executable.

If the problem persists, providing more details about your compilation process and the linker script could help in identifying the issue accurately.

**I prefer to verify the compilation from the disassembly**

vivekvpandya, 2024-01-17:
I run spike through debugger and see that I hit following condition
https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/ns16550.cc#L230
where reg_io_width is 1 and len is 8.
Is it something wrong with parameters passed to spike ?

scottj97, 2024-01-17:
There is a UART at 0x10000000. You'll need to either change your memory map to leave room for this device, or move the UART (by editing and recompiling Spike, or it looks like you can do it by providing your own device tree blob via `--dtb`).

vivekvpandya, 2024-01-18:
> There is a UART at 0x10000000. You'll need to either change your memory map to leave room for this device, or move the UART (by editing and recompiling Spike, or it looks like you can do it by providing your own device tree blob via `--dtb`).

Thanks @scottj97 ! Is there any conventional address range after which I should start my memory map?

scottj97, 2024-01-20:
I believe RISCV programs for bare metal typically start at 0x80000000.

Timmmm, 2024-06-05:
Ok after doing some reading it seems like `-m` won't help because Spike places some ROM & maybe MMIO devices in low addresses. Also based on some of the comments in [this thread](https://github.com/riscv-software-src/riscv-isa-sim/issues/145) it seems like you can't use an arbitrary entry point, but it must be above 0x80000000, maybe even *equal to* 0x80000000, I'm not sure.

svasekar, 2024-07-17:
HI Guys, I am trying to load elf, to SPIKE htif memory , by proving the path of elf file,  and not calling SPIKE , instead,  calling sim_t in my cpp wrapper. I get invalid address error while reading elf file.
Any idea on how to resolve it