riscv-software-src/riscv-isa-sim #1429: Misaligned Address Error in RISC-V Vector Extension Implementation using Spike Simulator

Myrausman, 2023-07-28:
We are currently working on a RISC-V vector extension implementation using GCC for our project. The C code compiles successfully with the vector extension options through the toolchain. However, we are encountering a "misaligned address" error during simulation with the Spike simulator.
```
#include <stddef.h>
#include <riscv_vector.h>

volatile uint32_t tohost __attribute__((section(".tohost")));
volatile uint32_t fromhost __attribute__((section(".fromhost")));

void saxpy(size_t n, const float a, const float *x, float *y) {
    size_t vl;
    vfloat32m8_t vx, vy;
    for (; n > 0; n -= vl) {
        vl = vsetvl_e32m8(n);
        vx = vle32_v_f32m8(x, vl);
        vy = vle32_v_f32m8(y, vl);
        vy = vfmsac_vf_f32m8(vy, a, vx, vl);
        vse32_v_f32m8(y, vy, vl);
        x += vl;
        y += vl;
    }

    // Signal completion to the "tohost" register after the loop
    tohost = 1;
}

int main() {
    // ... (rest of the code)

    saxpy(array_size, a, x, y);

    return 0;
}
```
If anyone has experience with RISC-V vector extensions or has encountered similar issues with Spike simulator, your expertise and guidance would be greatly appreciated.

Thank you for your support.

tianhu-zeng, 2023-08-10:
If you haven't added the "--misaligned" option to the command line, you can add it and try again.

tianhu-zeng, 2023-08-10:
> We are currently working on a RISC-V vector extension implementation using GCC for our project. The C code compiles successfully with the vector extension options through the toolchain. However, we [...]

If you haven't added the "--misaligned" option to the command line, you can add it and try again.

closed 2023-08-27 by Myrausman