riscv-software-src/riscv-isa-sim #1430: Issues in spike generated log file

SyedHassanUlHaq, 2023-07-31:
I am try to run a bare-metal program on spike for the Vector Extension
My configurations for the Toolchain is:
 ```
../configure --prefix=$HOME/rvv64 --with-arch=rv64gcv --with-abi=lp64 --enable-multilib
 ```
My configuration for spike is:
 ```
../configure --prefix=$HOME/rvv64 --with-varch=vlen:128,elen=32 --with-isa=rv64gcv --with-target=riscv64-unknown-elf 
 ```
this was my C code having vector instrinsics
```
#include <stddef.h>
#include <riscv_vector.h>

volatile uint64_t tohost __attribute__((section(".tohost")));
volatile uint64_t fromhost __attribute__((section(".fromhost")));

void saxpy(size_t n, const double a, const double *x, double *y) {
    size_t vl;
    vfloat64m8_t vx, vy;
    for (; n > 0; n -= vl) {
        vl = vsetvl_e64m8(n);
        vx = vle64_v_f64m8(x, vl);
        vy = vle64_v_f64m8(y, vl);
        vy = vfmsac_vf_f64m8(vy, a, vx, vl);
        vse64_v_f64m8(y, vy, vl);
        x += vl;
        y += vl;
    }

    // Signal completion to the "tohost" register after the loop
    tohost = 1;
}

int main() {
    const size_t array_size = 8;
    const double a = 2.0;
    double x[array_size];
    double y[array_size];

    // Initialize arrays x and y with data
    for (size_t i = 0; i < array_size; i++) {
        x[i] = (double)(i + 1);      // Values 1.0, 2.0, 3.0, ..., 8.0
        y[i] = (double)((i + 1) * 10); // Values 10.0, 20.0, 30.0, ..., 80.0
    }

    saxpy(array_size, a, x, y);

    return 0;
}

```

So I tried running the bare-metal program but spike in its log file always prints just the RV32I instructions with repetitive exception traps after 160 lines, tried PK too by coding the program accordingly but PK gets stuck with no output. I Have tried changing my C code multiple times but getting the same result out of them.

please help me through the issue

brucehoult, 2023-08-01:
Bare metal main has nowhere to return to, so you can't do that. You have to infinite loop after setting `tohost`, to give Spike time to see it -- it checks once every 10k instructions or so I think.

You're not showing how you built YOUR code. You can't just run a C main as the entry point. If nothing else, the stack is not set up, so you can't allocate local variables or call a function that allocates a stack frame. What is your `_start` code?

omerguzelelectronicguy, 2023-08-01:
> Bare metal main has nowhere to return to, so you can't do that. You have to infinite loop after setting `tohost`, to give Spike time to see it -- it checks once every 10k instructions or so I [...]

I also use bare metal mode and have an exit(I used the syscall startfile and other libraries from other open-source projects). I couldn't understand how the spike know the exit code. Can I write my exit region too? does spike understand from keyword _tohost or a specific address or infinity loop?

brucehoult, 2023-08-05:
Spike looks in your elf file for a section called ".tohost". The actual link address doesn't matter. If it exists, it checks the contents of that memory word every so often -- as I said, I think about once every 10000 emulated instructions.

closed 2023-08-22 by SyedHassanUlHaq