riscv-software-src/riscv-isa-sim #1431: Vector Mask instructions non zero vstart

ajayvreddy, 2023-08-01:
Hi,

I wanted to get clarification of the behavior of vector mask instructions when vstart is non-zero.

e.g. vcpop.m, the spec mentions:
`Traps on vcpop.m are always reported with a vstart of 0. The vcpop.m instruction will raise an illegal instruction exception if vstart is non-zero.`

Does it mean that on an illegal instruction exception, vstart should be written with value 0.
if yes, should it also set MSTATUS.VS to dirty because VStart was written.

aswaterman, 2023-08-01:
No, vstart shouldn't be written in this case.  And, as you said, vstart being changed should correspond to mstatus.VS being set to Dirty.

jrahmeh, 2023-08-01:
So illegal instruction traps should leave vstart as is, and all other traps should clear it?

aswaterman, 2023-08-01:
I don’t think that follows…?

aswaterman, 2023-08-01:
It’s an overly literal parsing of the spec. I think you know what it’s actually saying

jrahmeh, 2023-08-01:
I am not being pedantic, our team cannot agree on how to parse this.  One person suggested that what the spec is saying is: traps caused by vstart being non-zero should leave vstart alone.  All other traps should clear vstart.  Is that a reasonable reading of the spec?

aswaterman, 2023-08-01:
In general, instructions that result in illegal-instruction exceptions shouldn’t be modifying any state. The spec’s comment about traps being reported with a vstart of 0 is really about interruptibility.