riscv-software-src/riscv-isa-sim #1437: use SRET enter HS-mode, sstatus.SPIE sets 0

heiyuen1999, 2023-08-15:
[image]
This test program attempts to enter VS mode using SRET.According to the spec, sstatus.SPIE should be set to 1 after SRET.
And why is sepc cleared to 0?
[image]
The results of the spike run are as follows：
[image]
Thanks

scottj97, 2023-08-15:
Are you going to make me type all that in to recreate your problem? Why not paste the code as text instead of image? Help us help you.

heiyuen1999, 2023-08-16:
> Are you going to make me type all that in to recreate your problem? Why not paste the code as text instead of image? Help us help you.

Sorry.I've pasted the code as text.

scottj97, 2023-08-17:
I suspect what's happening here is that the interactive Spike debugger is seeing `vsstatus` when you ask for `sstatus`, since the core is in VS-mode. The actual `sstatus` does indeed have SPIE=1, which you can see by asking for `mstatus` instead. (`sstatus` is simply a masked version of `mstatus`).

Is this desirable behavior? Probably not...I think the interactive debugger should act as God and report all architectural CSRs as they are, instead of the virtualized versions when in VS-mode or VU-mode.

But I don't think there's any bug in Spike's execution of the target program. The only issue is with the interactive debugger reporting confusing values.

aswaterman, 2023-08-17:
I think you’re right, @scottj97. OTOH, we do benefit from being able to use the interactive mode to access memory under the current context’s address-translation regime, so we probably don’t want to make _all_ interactive commands behave as though they’re in god mode.

heiyuen1999, 2023-08-21:
> I think you’re right, @scottj97. OTOH, we do benefit from being able to use the interactive mode to access memory under the current context’s address-translation regime, so we probably don’t want to make _all_ interactive commands behave as though they’re in god mode.

Okay. I see. Thanks for the answer.

closed 2023-08-21 by heiyuen1999