riscv-software-src/riscv-isa-sim #1440: vector crypto insn vaeskf2.vi seems unmatch spec

GuoShibo-cn, 2023-08-17:
[image]
According to spike vaeskf2_vi function implementation, use "(round >> 1) - 1" to look up round constants, However, spec use 
[image]
(round >> 1) ,  could it be a slip of the pen?

aswaterman, 2023-08-17:
@kdockser any insight?

aamartin0000, 2023-08-17:
This was discussed and resolved here:
https://github.com/riscv/riscv-crypto/issues/336

Not sure if it's been updated everywhere.

kdockser, 2023-08-17:
The pseudocode in the latest version ([Vector Crypto Specification 1.0.0-rc2](https://github.com/riscv/riscv-crypto/releases/download/v/riscv-crypto-spec-vector.pdf)) has the correct equation
`aes_subword_fwd(aes_rotword(CurrentRoundKey[3])) XOR aes_decode_rcon((rnd>>1) - 1) XOR RoundKeyB[0];`

aswaterman, 2023-08-17:
So I think Spike is doing the right thing already.  Right, @GuoShibo-cn?

GuoShibo-cn, 2023-08-18:
Yes, Thanks, didn't realize spec change, spike is doing the right thing! It helps a lot.

closed 2023-08-18 by GuoShibo-cn