riscv-software-src/riscv-isa-sim #1441: Vector checks ordering

ajayvreddy, 2023-08-17:
I want to bring to attention the vector ordering checks for vector mask instructions (example:viota.m) and get your opinion on the right behavior.
in the file: riscv/insns/decode_macros.h
#define require_vector(alu) \
  do { \

    require_vector_vs; \

    require_extension('V'); \

    require(!P.VU.vill); \

    if (alu && !P.VU.vstart_alu) \

      require(P.VU.vstart->read() == 0); \

    WRITE_VSTATUS; \

    dirty_vs_state; \
  } while (0);

This macro dirties the mstatus.vs : `dirty_vs_state` if the above conditions are met. (and if any condition is not met, it raises illegal exception)

But for vector mask instructions (viota.m: riscv/insns/viota_m.h), we can still raise illegal exception even later, such as  when vector register group overlaps V0. 

In the above mentioned case, we would see a illegal exception but the mstatus.vs would be modified.

I think Spike should not change mstatus.vs and correct this behavior.

aswaterman, 2023-08-17:
It’s harmless, I think, but I agree it would be preferable to perform overlap checks before dirtying VS. This is also more likely to match practical HW implementations.

ajayvreddy, 2023-08-18:
Okay. Thanks for the confirmation.