riscv-software-src/riscv-isa-sim #1444: question of riscv/insns/vmv_s_x.h code

leiyu668, 2023-08-25:
I don't think vmv.s.x should update vd[0] element when vstart=1 and vl =2, but code in riscv/insns/vmv_s_x.h did, is it correct?

aswaterman, 2023-08-25:
It’s a fair question. This instruction would never set vstart=1, even if interrupted. Therefore, trapping is a legal option (but is not required). I would say it is UNSPECIFIED, so writing vd[0], trapping, or NOP would all be reasonable choices. But this is just an opinion.

leiyu668, 2023-08-25:
I can build this scenario with the following code:
li t0, 2
vsetvli t1, t0, e8,m1,ta,ma
li   t0,1
csrw   vstart,t0
vmv.s.x v1, s0

aswaterman, 2023-08-25:
Thank you for the test case. At this point, I think it is an ISA question, not a Spike question.

leiyu668, 2023-08-25:
Can't agree with you more, aswaterman.
Thank you a lot for your kind comments. Have a good day!

aswaterman, 2023-08-25:
You as well!

closed 2023-09-23 by EventScheduler