riscv-software-src/riscv-isa-sim #1447: Are G-stage D bits set correctly as a result of setting a VS-stage D bit?

aswaterman, 2023-08-26:
Consider the following scenario: an explicit store to a clean page causes a VS-stage D bit to be set.  That VS-stage page table lives on a G-stage page which is itself clean.  So, we need to set the D bit on the G-stage PTE corresponding to the GPA of the VS-stage page table.

I'm not seeing how that's happening.  The way we set the VS-stage D bit is by first obtaining the SPA of the VS-stage PTE here https://github.com/riscv-software-src/riscv-isa-sim/blob/5854ab52181f96723178f903e623d21e61f4fe61/riscv/mmu.cc#L506; note we call `s2xlate` with `LOAD` permissions.  And then we directly write to that SPA here https://github.com/riscv-software-src/riscv-isa-sim/blob/5854ab52181f96723178f903e623d21e61f4fe61/riscv/mmu.cc#L540.  Neither of those actions appear to cause the D bit to be set for the G-stage PTE corresponding to the GPA of the VS-stage page table.  If I'm right, a possible fix might be to insert the statement `s2xlate(addr, base + idx * vm.ptesize, STORE, type, virt, false);` just before the call to `pte_store`.  (We don't care about the return value of `s2xlate` in this case; if it succeeds, we already know the return value will equal `pte_paddr`.)

cc @adurbin-rivos @ved-rivos

ved-rivos, 2023-08-26:
I see a few additional issues in the implementation a) the `pte_store` is not atomic b) it does not factor in that the `pte` may change between the first load and the attempt to set the a and/or d bit (e.g. a super page being demoted). I will send a pr with a proposal to fix.

aswaterman, 2023-08-26:
The non-atomicity is not a bug because Spike is single-threaded.  This assumption is made throughout Spike (even AMOs are implemented without using atomics).

ved-rivos, 2023-08-26:
I see. Then I agree that your proposed fix to invoke `s2xlate` suffices.

aswaterman, 2023-08-27:
Cool.  Mind making the PR?  (I'd do it myself but I'm on my phone right now)

ved-rivos, 2023-08-27:
I started making the PR but then I also noticed that the implementation seems to always set tinst set to 0 which I believe is not allowed since the trap does report non-zero value in tval2 and so wanted to fix that. This could be done by introducing another "implicit/explicit" parameter to s2xlate or by expanding the access_type to indicate if its a IMPLICIT_VS_PT_LOAD or a IMPLICIT_VS_PT_STORE. The second leads to more touches - such as to the PMP checks - so I am leaning to first. Please tell me if there is a preference.

ved-rivos, 2023-08-27:
Sent PR #1448