riscv-software-src/riscv-isa-sim #1449: how to generate compressed instructions regression test

GMYMingyu, 2023-08-28:
Hi,
1）Objective：I would like to generate compressed instruction regression test with riscv-dv, which means the amount of compressed instrucions is large, exceeding 50% of all instructions
2)  My action: I set a new test name in the yaml: +instr_cnt=1000 +disable_compressed_instr=0 +num_of_sub_program=0, accoding to the Section Runtime options of the generator on https://htmlpreview.github.io/?https://github.com/google/riscv-dv/blob/master/docs/build/singlehtml/index.html#runtime-options-of-the-generator. However, the generated .S file does not list any compressed instruction.
Thanks

GMYMingyu, 2023-08-28:
In my expectation, set +disable_compressed_instr=0 could generate multiple compressed instructions in the generated  .s file, riscv-dv randomly generating compressed instructions.

scottj97, 2023-08-28:
This is the ISA simulator Spike. You are looking for some other project (perhaps [riscv-dv](https://github.com/chipsalliance/riscv-dv)?)

closed 2023-08-28 by scottj97