riscv-software-src/riscv-isa-sim #1451: Spike load instructions bigger than 32 bits for not align address

trambu, 2023-08-31:
Let's assume that we have 0x80001707 in the `ra` register, and then we call the `ret` instruction. Also, assume that the memory content(code memory) is as follows:
```
800016fc:	f19ff0ef          	jal	ra,80001614 <thread_entry>
80001700:	4581                	li	a1,0
80001702:	4501                	li	a0,0
80001704:	945ff0ef          	jal	ra,80001048 <main>
80001708:	87aa                	mv	a5,a0
8000170a:	fcf42823          	sw	a5,-48(s0)
8000170e:	fc943c23          	sd	s1,-40(s0)
```
In this case, the spike will jump to the `0x80001706` address, but the instruction that will load is `0x282387aa945f`, which includes 6 bytes(48bits for instruction) value from `0x80001706` and `0x80001708` and `0x8000170a` addresses. You can see the spike log in `1.log`.

Tested on spike commit: `5854ab52181f96723178f903e623d21e61f4fe61` However, it is reproducible to all available spike versions.

[1.log](https://github.com/riscv-software-src/riscv-isa-sim/files/12486052/1.log)

scottj97, 2023-08-31:
I agree with Andrew. The 48-bit fetch has nothing to do with the misaligned `ra`; it's because the instruction decodes to a 48-bit length. I don't see any issue here?

trambu, 2023-09-01:
> What’s the problem? Ignoring the LSB is correct, and supporting instruction fetch of 6-byte-long instructions is valid implementation-defined behavior.

Thank you for your response.
Does the `rv64imafdc` ISA support instruction fetching of 6-byte-long and 8-byte-long? I would appreciate it if you could send me the reference because I could not find it.

aswaterman, 2023-09-01:
The intro chapter in the unprivileged ISA manual says that ILEN can be greater than 32.  Note that there are currently no _standard_ extensions with instructions longer than 32 bits, so longer instructions are inherently custom extensions at this time.  Hence, the specific choice of encoding for 6-byte and 8-byte instructions is Spike-specific.

trambu, 2023-09-06:
Thank you @aswaterman for your reply, so I will close the issue.

closed 2023-09-06 by trambu