riscv-software-src/riscv-isa-sim #1455: Access Memory Abstract Command

abhi-valtrix, 2023-09-07:
Need some help with the access memory command : 

I am trying to write a valid memory location with the following parameters :

Open On-Chip Debugger - poll off

> riscv dmi_read 0x16       
0x2000002

> riscv dmi_read 0x5              - this has the memory location i am writing to.
0x43a7e000

> riscv dmi_read 0x4             - value
0x4d2

> riscv dmi_write  0x17 0x2210000    --- > the fields' values are mentioned below

CMDTYPE   	    :2
AAMVIRTUAL     :0
AAMSIZE            :2
AAMPOSTINCREMENT     :0
WRITE                                :1
TARGET_SPECIFIC           :0

ABSTRACTCS ( below ) returns cmderr as 2 - unsupported - am I missing something here ? or is the command not supported ?
> riscv dmi_read 0x16 
0x2000202

Appreciate any help I can get on this :+1:

timsifive, 2023-09-07:
I thought I had implemented this, but apparently not.

If you look in debug_module.cc, perform_abstract_command(), you'll see that only abstract register access is supported.

abhi-valtrix, 2023-09-07:
Thanks @timsifive !

closed 2023-09-07 by abhi-valtrix