riscv-software-src/riscv-isa-sim #1460: How to interrupt CPU under WFI

rahul-neubla, 2023-09-13:
Hi,

I am using Spike for CPU simulation running along with my Coprocessor simulator. To sync between CPU and coprocessor, I want to simulate interrupt from coprocessor to CPU, with CPU waiting in WFI. Could I simply rely on updating `in_wfi` in `processor.cc` or is there some elegant way to do this. Any help/pointers would be very useful :)

Thanks,
Rahul

jerryz123, 2023-09-13:
Updating `in_wfi` is insufficient, as a processor can take an interrupt while not in a `wfi` state. 
Updating `state.mip` is the correct approach. You can take a look at how the CLINT/PLIC raise MIP in clint.cc and plic.cc for examples.

There is also an example of how to use the Spike C++ API for co-simulation here: https://github.com/ucb-bar/testchipip/blob/master/src/main/resources/testchipip/csrc/cospike_impl.cc

rahul-neubla, 2023-09-14:
> Updating `in_wfi` is insufficient, as a processor can take an interrupt while not in a `wfi` state. Updating `state.mip` is the correct approach. You can take a look at how the CLINT/PLIC raise MIP [...]

Thanks, thats quite helpful

closed 2023-09-14 by rahul-neubla