riscv-software-src/riscv-isa-sim #1461: Wrong Underflow float exceptions fflags in some cases of fcvt, fdiv, fmul and fmadd instructions in spike

Shanchuang666, 2023-09-14:
In Section 11.4 of the risc-v Unprivileged ISA spec (v20190309), it is stated to detect tininess after rounding, but the following cases I tested got a different floating-point exception fflags in spike.

```
fdiv:
    csrwi frm, 0
    csrwi fflags, 0
    fdiv.s f26, f14, f12      #(0_00000001_11111111111111111111111) / 0x40000000

    csrwi frm, 1
    csrwi fflags, 0
    fdiv.s f27, f14, f12      #(0_00000001_11111111111111111111111) / 0x40000000

fmul:
    csrwi frm, 0
    csrwi fflags, 0
    fmul.d f26, f17, f16     # 0x3ff0000000000001 * 0x000fffffffffffff
   
    csrwi frm, 1
    csrwi fflags, 0
    fmul.d f27, f17, f16     # 0x3ff0000000000001 * 0x000fffffffffffff
    
    csrwi frm, 0
    csrwi fflags, 0
    fmul.d f28, f20, f21     # 0x0010000000000000 * 0x3fefffffffffffff

    csrwi frm, 1
    csrwi fflags, 0
    fmul.d f29, f20, f21     # 0x0010000000000000 * 0x3fefffffffffffff

fma:
    csrwi frm, 0
    csrwi fflags, 0
    fmadd.d f26, f17, f16, f19   #(0x3ff0000000000001 * 0x000fffffffffffff) + (+0)

    csrwi frm, 1
    csrwi fflags, 0
    fmadd.d f27, f17, f16, f19   #(0x3ff0000000000001 * 0x000fffffffffffff) + (+0)

fcvt:
    csrwi frm, 0
    csrwi fflags, 0
    fcvt.s.d       f26, f18    	# 0x380FFFFFFFFFFFFF

    csrwi frm, 1
    csrwi fflags, 0
    fcvt.s.d       f27, f18    	# 0x380FFFFFFFFFFFFF
```

```
core   0: >>>>  fdiv
core   0: 0x000000008000014c (0x00205073) csrwi   frm, 0
core   0: 3 0x000000008000014c (0x00205073) c2_frm 0x0000000000000000
core   0: 0x0000000080000150 (0x00105073) csrwi   fflags, 0
core   0: 3 0x0000000080000150 (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x0000000080000154 (0x18c77d53) fdiv.s  fs10, fa4, fa2
core   0: 3 0x0000000080000154 (0x18c77d53) c1_fflags 0x0000000000000003 f26 0xffffffff00800000
core   0: 0x0000000080000158 (0x0020d073) csrwi   frm, 1
core   0: 3 0x0000000080000158 (0x0020d073) c2_frm 0x0000000000000001
core   0: 0x000000008000015c (0x00105073) csrwi   fflags, 0
core   0: 3 0x000000008000015c (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x0000000080000160 (0x18c77dd3) fdiv.s  fs11, fa4, fa2
core   0: 3 0x0000000080000160 (0x18c77dd3) c1_fflags 0x0000000000000003 f27 0xffffffff007fffff
core   0: >>>>  fmul
core   0: 0x0000000080000104 (0x00205073) csrwi   frm, 0
core   0: 3 0x0000000080000104 (0x00205073) c2_frm 0x0000000000000000
core   0: 0x0000000080000108 (0x00105073) csrwi   fflags, 0
core   0: 3 0x0000000080000108 (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x000000008000010c (0x1308fd53) fmul.d  fs10, fa7, fa6
core   0: 3 0x000000008000010c (0x1308fd53) c1_fflags 0x0000000000000001 f26 0x0010000000000000
core   0: 0x0000000080000110 (0x0020d073) csrwi   frm, 1
core   0: 3 0x0000000080000110 (0x0020d073) c2_frm 0x0000000000000001
core   0: 0x0000000080000114 (0x00105073) csrwi   fflags, 0
core   0: 3 0x0000000080000114 (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x0000000080000118 (0x1308fdd3) fmul.d  fs11, fa7, fa6
core   0: 3 0x0000000080000118 (0x1308fdd3) c1_fflags 0x0000000000000003 f27 0x000fffffffffffff
core   0: 0x000000008000011c (0x00205073) csrwi   frm, 0
core   0: 3 0x000000008000011c (0x00205073) c2_frm 0x0000000000000000
core   0: 0x0000000080000120 (0x00105073) csrwi   fflags, 0
core   0: 3 0x0000000080000120 (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x0000000080000124 (0x135a7e53) fmul.d  ft8, fs4, fs5
core   0: 3 0x0000000080000124 (0x135a7e53) c1_fflags 0x0000000000000003 f28 0x0010000000000000
core   0: 0x0000000080000128 (0x0020d073) csrwi   frm, 1
core   0: 3 0x0000000080000128 (0x0020d073) c2_frm 0x0000000000000001
core   0: 0x000000008000012c (0x00105073) csrwi   fflags, 0
core   0: 3 0x000000008000012c (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x0000000080000130 (0x135a7ed3) fmul.d  ft9, fs4, fs5
core   0: 3 0x0000000080000130 (0x135a7ed3) c1_fflags 0x0000000000000003 f29 0x000fffffffffffff
core   0: >>>>  fma
core   0: 0x0000000080000134 (0x00205073) csrwi   frm, 0
core   0: 3 0x0000000080000134 (0x00205073) c2_frm 0x0000000000000000
core   0: 0x0000000080000138 (0x00105073) csrwi   fflags, 0
core   0: 3 0x0000000080000138 (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x000000008000013c (0x9b08fd43) fmadd.d fs10, fa7, fa6, fs3
core   0: 3 0x000000008000013c (0x9b08fd43) c1_fflags 0x0000000000000001 f26 0x0010000000000000
core   0: 0x0000000080000140 (0x0020d073) csrwi   frm, 1
core   0: 3 0x0000000080000140 (0x0020d073) c2_frm 0x0000000000000001
core   0: 0x0000000080000144 (0x00105073) csrwi   fflags, 0
core   0: 3 0x0000000080000144 (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x0000000080000148 (0x9b08fdc3) fmadd.d fs11, fa7, fa6, fs3
core   0: 3 0x0000000080000148 (0x9b08fdc3) c1_fflags 0x0000000000000003 f27 0x000fffffffffffff
core   0: >>>>  fcvt
core   0: 0x0000000080000164 (0x00205073) csrwi   frm, 0
core   0: 3 0x0000000080000164 (0x00205073) c2_frm 0x0000000000000000
core   0: 0x0000000080000168 (0x00105073) csrwi   fflags, 0
core   0: 3 0x0000000080000168 (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x000000008000016c (0x40197d53) fcvt.s.d fs10, fs2
core   0: 3 0x000000008000016c (0x40197d53) c1_fflags 0x0000000000000001 f26 0xffffffff00800000
core   0: 0x0000000080000170 (0x0020d073) csrwi   frm, 1
core   0: 3 0x0000000080000170 (0x0020d073) c2_frm 0x0000000000000001
core   0: 0x0000000080000174 (0x00105073) csrwi   fflags, 0
core   0: 3 0x0000000080000174 (0x00105073) c1_fflags 0x0000000000000000
core   0: 0x0000000080000178 (0x40197dd3) fcvt.s.d fs11, fs2
core   0: 3 0x0000000080000178 (0x40197dd3) c1_fflags 0x0000000000000003 f27 0xffffffff007fffff
```

aswaterman, 2023-09-14:
Sorry if this sounds condescending, but this is the first GitHub issue you've filed.  Are you absolutely certain this is a bug and not a misunderstanding of the FP standard?

Shanchuang666, 2023-09-15:
Sorry, let me explain this in more detail.

[image]

[image]

From IEEE754-2008 standard, it shall  detect tininess in the same way for all operations.
From “F” Standard Extension for Single-Precision Floating-Point, Version 2.2, it shall detect tininess after rounding.

In this multiplication case, the vd is 0x0010000000000000 which is a minimum normal result. But both underflow and inexact exception flags are raised.
```
csrwi frm, 0
csrwi fflags, 0
fmul.d f28, f20, f21     # 0x0010000000000000 * 0x3fefffffffffffff
```
core   0: 0x0000000080000124 (0x135a7e53) fmul.d  ft8, fs4, fs5
core   0: 3 0x0000000080000124 (0x135a7e53)  c1_**fflags** 0x000000000000000**3**  f28 0x**0010000000000000**

In this multiplication case, the vd is 0x0010000000000000 which is a minimum normal result, and the spike only raises an inexact flag.
```
csrwi frm, 0
csrwi fflags, 0
fmul.d f26, f17, f16     # 0x3ff0000000000001 * 0x000fffffffffffff
```
core   0: 0x000000008000010c (0x1308fd53) fmul.d  fs10, fa7, fa6
core   0: 3 0x000000008000010c (0x1308fd53)  c1_**fflags** 0x000000000000000**1**  f26 0x**0010000000000000**

In this division case, the vd is 0x00800000 which is a minimum normal result. But both underflow and inexact exception flags are raised.
```
csrwi frm, 0
csrwi fflags, 0
fdiv.s f26, f14, f12      # 0x00ffffff / 0x40000000
```
core   0: 0x0000000080000154 (0x18c77d53) fdiv.s  fs10, fa4, fa2
core   0: 3 0x0000000080000154 (0x18c77d53)  c1_**fflags** 0x000000000000000**3**  f26 0xffffffff**00800000**

Shanchuang666, 2023-09-15:
> Sorry if this sounds condescending, but this is the first GitHub issue you've filed. Are you absolutely certain this is a bug and not a misunderstanding of the FP standard?

Thank you very much for your review. Please kindly refer to my follow-up explanation.

aamartin0000, 2023-09-15:
If I may, please take a look at this: http://www.jhauser.us/arithmetic/SoftFloat-3/doc/SoftFloat-FAQ.html
The second question may provide some insight into how the Underflow flag is calculated.
(I believe Spike implements FP using SoftFloat.)

Shanchuang666, 2023-10-08:
> If I may, please take a look at this: http://www.jhauser.us/arithmetic/SoftFloat-3/doc/SoftFloat-FAQ.html The second question may provide some insight into how the Underflow flag is calculated. (I believe Spike implements FP using SoftFloat.)

I got it, thanks!