riscv-software-src/riscv-isa-sim #1462: Need clarification regarding program buffer execution.

abhi-valtrix, 2023-09-15:
I am unable to see any change post the program buffer execution. Am I missing something here ? Appreciate any help :)

opcode :	00468693         addi   a3,a3,4

openocd commands : 

> poll off
> reg 13              
a3 (/32): 0x00000000

> riscv dmi_write 0x20    0x00468693                    - writing progbuf0 with the opcode of the addi instruction.
> riscv dmi_write 0x21    0x00468693                    - writing progbuf1 with the opcode of the addi instruction. 
> reg 13                         
a3 (/32): 0x00000000

> riscv dmi_write  0x17 0x40000                         -  only postexec bit is 1
> reg 13                        
a3 (/32): 0x00000000

> riscv dmi_read 0x16
0x2000002

timsifive, 2023-09-15:
OpenOCD will cache the value of all GPRs. I don't think there's code to realize that dmi_write can change their values.

abhi-valtrix, 2023-09-18:
Thanks @timsifive  was able to bypass the cache using the force option. Closing the issue.

closed 2023-09-18 by abhi-valtrix