riscv-software-src/riscv-isa-sim #1464: Request for RV32IV Core Testing Support in Spike

latifbhatti, 2023-09-18:
We are interested in testing our RV32IV core in the Spike simulator. Can you please provide support or guidance for this? Your assistance would be highly valuable to our project.
repository link:
https://github.com/merledu/vaquita

ekliptik, 2023-11-01:
Spike isn't a CPU core simulator or an HDL simulator, it is an ISA-level simulator. You can't test your CPU core *in* spike but you can test it *against* spike by for example compiling C programs with CSmith and running them in your simulated/FPGA-emulated CPU core as well as running them in spike and checking that the results match.

GitHub issues aren't a good place to look for consultancy services and similar. The commonly intended use of issues is to report bugs and suggest improvements to the code at hand. As your request doesn't match this use case, I suggest closing this issue, and looking for open community discussion spots like general email threads and RISC-V related forums.

jerryz123, 2024-01-17:
The spike C++ API can be used in a tandem verification setup for vector units, as demonstrated here https://github.com/ucb-bar/testchipip/blob/master/src/main/resources/testchipip/csrc/cospike_impl.cc

closed 2024-01-17 by jerryz123