riscv-software-src/riscv-isa-sim #1465: After adding custom instructions, a memory leak occurs when running spike

Lebattt, 2023-09-19:
Hi,
In disasm.cc, I added a custom directive modeled after the csr directive:
`add_insn(new disasm_insn_t("csrrwx", match_csrrwx, mask_csrrwa, {&xrd, &csr, &zimm5}));`

But after spike execution exits, an error occurs: free(): invalid next size (fast)

Could you help me with that?

ekliptik, 2023-11-01:
I suggest creating a fork here on github so that spike contributors could take a look at your code and reproduce your issue. My personal suggestion would be to use a debugger and sanitizers as a part of your development flow. That being said, this is an issue with your code, rather than this existing repository - it's not a bug report or a feature requests. Maybe you could strike up a conversation in a more open community discussion place like the (SiFive forum)[https://forums.sifive.com/] instead of keeping this issue open

Lebattt, 2023-11-02:
> I suggest creating a fork here on github so that spike contributors could take a look at your code and reproduce your issue. My personal suggestion would be to use a debugger and sanitizers as a [...]

closed 2023-11-02 by Lebattt

Lebattt, 2023-11-02:
Thank you for your reply.

I have solved the issue. There are some custom registers defined with very long character and should be abbreviated.