riscv-software-src/riscv-isa-sim #1472: isa flag caused to run infinity loop.

omerguzelelectronicguy, 2023-09-27:
I am running spike with two different commands
`spike -l --log-commits                      -d --debug-cmd=spike.script --log=spike_trace.log  cm.bare.riscv`
`spike -l --log-commits --isa=rv64gc -d --debug-cmd=spike.script --log=spike_trace.log  cm.bare.riscv`
The second one was never finished. The first one is finished as expected. I'd like to know about the changing behaviour when I gave the ISA option to spike. I assumed the default isa option for spike is rv64gc when we don't supply.
The program is a coremark executable with minor changes.
Could anyone help me to understand this?

omerguzelelectronicguy, 2023-09-28:
I found that the default isa is `--isa=RV64IMAFDC_zicntr_zihpm`.  I didn't know that we should give the `base counter and timers` as `zicntr` and the `hardware performance counter` as `zihpm`. So the problem is the time count I think.

omerguzelelectronicguy, 2023-10-10:
I tried to standardise my compilation and simulation flow. However, `RV64IMAFDC_zicntr_zihpm` or `rv64imafdc_zicntr_zihpm` is not allowed by gcc. Additionally, gcc requires `zicsr` when I gave it `rv64imafdc_zicntr_zihpm` instead of `rv64gc`. Isa flag of gcc and spike is not matched as I understand. Am I wrong? What am I missing?

ekliptik, 2023-11-01:
The first thing that could help you understand misbehaving programs is by running spike with a [debugger](https://github.com/riscv-software-src/riscv-isa-sim#debugging-with-gdb). Since your issue was with missing extensions, you were trying to execute instructions that spike considered invalid, so you'd probably see Spike try and enter a trap handler for an invalid opcode trap.

With gcc, you can select the architecture extensions. You should always set the ISA for a compiler explicitly unless you're using a "host compiler" that is suppoed to be guaranteed to be native with your system. Look into `-march`, `-mcpu`, `-mabi` flags for gcc and clang. Also, the defaults of gcc and spike aren't really required to match, they are separate projects

omerguzelelectronicguy, 2023-11-03:
Thank you for a clear answer. Since this is one of the most common simulators for RISCV, I think making the usage of spike and GCC matched will be helpful

closed 2023-11-03 by omerguzelelectronicguy

ekliptik, 2023-11-03:
Now that I reread your messages, it looks like I misunderstood your issue. I was assuming you were relying on default features also in GCC.

About GCC bringing in Zicsr when using Zicntr: this is helpful, the Zicntr does use CSR, so it's kind of nice, but it could be an error to not have Zicsr when using Zicntr instead, so that the user is aware they're not getting exactly what they asked for. As far as the behavior of spike, since the ISA specification technically declares no requirement to have Zcsr when implementing Zicntr, I don't think it should match current gcc's behavior, since it's supposed to be really 1:1 with the spec, I think.

About default ISA values:
While it would be nice to have defaults work across gcc and simulators, it's a rare use case among embedded engineers, who are used to controlling everything to make sure they don't rely on any bad or changing defaults. It would lower the bar for newcomers used to native programming but also be rater misleading, implying that default architectures are reasonable to rely upon when cross-compiling.
Also, there is no single GCC. There is one GCC project, but when building GCC, you can influence default behavior like this. And then there are forks with vendor-specific patches applied.