riscv-software-src/riscv-isa-sim #1478: Changing PMP Grain through CLI

iamKarthikBK, 2023-10-11:
Hi

I'm aware of the option to change the number of PMP entries using the CLI option ``--pmpregions=<n>``. I think it would be helpful if we could also set the PMP grain in a similar fashion. I want to use spike with a pmp grain other than the default value. I went through the code base a little bit and found [this](https://github.com/riscv-software-src/riscv-isa-sim/blob/b5d13f36053d08e310da01e92e8ebc2d29623483/riscv/processor.cc#L669) function. Can we expose it to be called through the CLI ? Perhaps an option ``--pmpgrain=<n>`` ?

Writing a grain that is less than the minimum would simply be ignored based on the current framework, and all of that seems right. I would appreciate it if someone can throw some more light on this. As of this moment, in order to change the PMP grain, i'd have to do [this](https://github.com/riscv-software-src/riscv-isa-sim/issues/814) and then rebuild spike. Please correct me if I'm wrong. Thanks in advance!

iamKarthikBK, 2023-10-11:
If this is something the maintainers would like, I can work on this feature addition.

jerryz123, 2023-10-17:
I think this flag would be ok to add. 

Nit: it should be 'pmpgranularity'.

iamKarthikBK, 2023-10-28:
Thanks for the comment, @jerryz123 .
Do you think my PR (#1490) does the trick?
Please let me know if I've missed something, and I'll be happy to make the change.

Changes in this PR have been compile-tested by me.

PR #1490 (2023-11-09): Add support for setting PMP Granularity over CLI
Resolves #1478
closed 2023-11-09 by aswaterman