riscv-software-src/riscv-isa-sim #1485: How to run cm.jalt instruction of Zcmt Extension in Embench-iot programs by Spike?

gakubochama, 2023-10-18:
I tried to run cm.jalt instruction of Zcmt Extension in Embench-iot programs by Spike, but I can't.

`spike --isa=rv32ima_zicsr_zca_zcb_zcmp_zcmt -d cubic`
(cubic is elf file)

the console output↓
`core   0: 0x8000003e (0x00004f81) c.li    t6, 0
(spike) 
core   0: 0x80000040 (0x80012137) lui     sp, 0x80012
(spike) 
core   0: 0x80000044 (0x0000a1a6) cm.jalt 105
core   0: exception trap_instruction_access_fault, epc 0x80000044
core   0:           tval 0x000001a4
(spike) 
core   0: exception trap_instruction_access_fault, epc 0x00000000
core   0:           tval 0x00000000
(spike) 
core   0: exception trap_instruction_access_fault, epc 0x00000000
core   0:           tval 0x00000000`

[Zc's specification](https://github.com/riscv/riscv-code-size-reduction/tree/main) says
> Table jump uses a 256-entry XLEN wide table in instruction memory to contain function addresses.
The table must be a minimum of 64-byte aligned.

I could run Zca,Zcb,Zcmp instructions by Spike without any problems.

aswaterman, 2023-10-18:
The jvt CSR has not been initialized to a value that points to a useful table. Note that the exception address 0x1a4 is equal to the cm.jalt target of 105, times the table entry size of 4. 

This isn’t a Spike issue; you should consult the Zcmt spec and your toolchain documentation for additional info.

closed 2023-10-18 by aswaterman

jnk0le, 2023-10-20:
>so where is the address of table in instruction memory??

Usually, the table should be generated by compiler + some linker symbols
You could also manually put function pointers there

gakubochama, 2023-10-20:
@jnk0le 
Ok,manually...
But, why wasn't the table created automatically by the compiler, perhaps because the start function, which calls the main function, was written directly in assembly language?