riscv-software-src/riscv-isa-sim #1488: RISCV32 error in generating instruction type

Pruthvi-Parate, 2023-10-20:
I used this command to compile : riscv32-unkown-elf-gcc -march=rv32i -mabi=ilp32 -ffreestanding -nostdlib -o project.c 
riscv32-unknown-elf-objdump -d -r out > project.txt
   10100:	00178793          	add	a5,a5,1
Instead of getting I type instruction I am getting R type instruction in riscv32 gnu toolchain. 
[assembly.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/13056579/assembly.txt)

ekliptik, 2023-11-01:
Assembly is not machine code. When you write assembly, you can use an immediate operand in an `add` instruction. It will turn into `addi`, as if `add` was a pseudo-instruction. Similarly, objdump will disassemble `addi` as `add` since it is clear to the reader that for example 5 is an immediate and not a register. Compiling assembly will also compress instructions when possible, and correspondingly show their uncompressed opcode for readability. 

To really see "true machine code", try adding `-M no-aliases` to your objdump invocation.

Pruthvi-Parate, 2023-11-02:
Ok But if I use previous version of compiler it is working.

ekliptik, 2023-11-02:
Well then, which versions

Pruthvi-Parate, 2023-11-03:
riscv 11

ekliptik, 2023-11-03:
I meant both versions. What version shows you `addi`, what version shows you `add`? Where did you install / download them from? Are you testing this like `objdump -d a.out`? Without this information for reproducing it, there's no action for contributors to take to figure out your problem.

Pruthvi-Parate, 2023-11-05:
Ok when I used  riscv32-unkown-elf-gcc -march=rv32i -mabi=ilp32 -ffreestanding -nostdlib -o project.c then it gave 'add' and when I used riscv64-unkown-elf-gcc -march=rv32i -mabi=ilp32 -ffreestanding -nostdlib -o project.c then it gave addi , I downloaded them from official github repository riscv-gnu-toolchain. I am using this command to objdump :  riscv64-unknown-elf-objdump -d -r out > project.txt

ekliptik, 2023-11-06:
Were riscv32-unkown-elf-gcc and riscv64-unkown-elf-gcc both the same version? Which exact version? Can you share project.c, ideally reduced to a single small function that demonstrates the inconsistency you are talking about?

Pruthvi-Parate, 2023-11-06:
riscv32 version: 13.2.0, riscv64 version: 8.3.0 , Here is my project.c file
[project.txt](https://github.com/riscv-software-src/riscv-isa-sim/files/13267361/project.txt)

ekliptik, 2023-11-06:
I have reproduced what you are describing:

```
> ./xpack-riscv-none-elf-gcc-13.2.0-2/bin/riscv-none-elf-gcc ~/donut.c -c -w && ./xpack-riscv-none-elf-gcc-13.2.0-2/bin/riscv-none-elf-objdump donut.o -d | grep add | head -n 1
   0:	d4010113          	add	sp,sp,-704
> ./xpack-riscv-none-embed-gcc-8.3.0-2.3/bin/riscv-none-embed-gcc ~/donut.c -c -w && ./xpack-riscv-none-embed-gcc-8.3.0-2.3/bin/riscv-none-embed-objdump donut.o -d | grep add | head -n 1
   0:	d4010113          	addi	sp,sp,-704
```

Please notice that both instructions are encoded as d4010113 so the *actual machine instructions are exactly equal*. There has apparently been a change in the behavior of GNU objdump between version 8 and 13 that for the RISC-V target. What's interesting is that already the version 8 is reporting `add` for compressed `c.add`, so already version 8 was doing some readability expansion. If you'd like to find the commit that caused this change, you might want to take a look at the git log / git blame of opcodes/riscv-opc.c in the binutils project. However, this isn't related to this simulator repository, so I suggest you close this issue.

closed 2024-03-15 by Pruthvi-Parate