riscv-software-src/riscv-isa-sim #1493: stdout and stderr can not be redirected

ekliptik, 2023-11-01:
When printf (implemented with HTIF) is used in an executable run with spike, the printed text can not be redirected. As an example, `spike --isa=RV32GC whatever > /dev/null 2>&1` still prints something. This complicates the usage of spike for automated testing.

My guess is that the file descriptor duplication in `fesvr/syscall.cc:173` is somehow at fault.

aswaterman, 2023-11-01:
I'm not able to replicate the behavior.  With a simple test case, at least, I see that redirecting stdout suffices.

```
$ cat test.c
#include <stdio.h>
int main()
{
  printf("hello world\n");
  return 0;
}
$ spike pk ./a.out
bbl loader
hello world
$ spike pk ./a.out 2> /dev/null
bbl loader
hello world
$ spike pk ./a.out > /dev/null
$ spike pk ./a.out > /dev/null 2>&1
$
```

ekliptik, 2023-11-02:
Interesting. Here's my reproducer:
[repro.tar.gz](https://github.com/riscv-software-src/riscv-isa-sim/files/13239211/repro.tar.gz)
Let `cc` be clang or gcc of your choice
`cc scream.c htif.c a.s -nostdlib -Tvanilla-spike.ld -march=rv32g`
`spike --isa=RV32G a.out > /dev/null` will not surpress output