riscv-software-src/riscv-isa-sim #1497: Question about the writability of henvcfg.pbmte when menvcfg.pbmte is clear

viktoryou, 2023-11-09:
Hi,
According to the Priv.1.12 spec (Chapter 3.1.18)
>for implementations with the hypervisor extension, henvcfg.PBMTE is read-only zero if menvcfg.PBMTE is zero.

I found the behaviour of spike only satisfies "henvcfg.PBMTE reads as zero when menvcfg.PBMTE is zero". 
In a situation like，

step 1: set menvcfg.pbmte=0;
step 2: set henvcfg.pbmte=1; 
step 3: set menvcfg.pbmte=1;
step 4: read henvcfg;

I think the expected henvcfg.pbmte bit should be 0, while the outcome of spike is 1. I think the description above in the spec is clear and the spike violates the rule. Please help furthur check.

scottj97, 2023-11-10:
It's not clear to be exactly what the behavior should be here. What about the following scenario:

1. Set menvcfg.pbmte=1
2. Set henvcfg.pbmte=1
3. Set menvcfg.pbmte=0 (so henvcfg.pbmte now reads as 0)
4. Set menvcfg.pbmte=1
5. Read henvcfg.pbmte

Should henvcfg.pbmte read the old value of 1? Or was it cleared to 0 by one of the steps above?

viktoryou, 2023-11-11:
My test yielded the result that "henvcfg.pbmte" was reading the old value of 1. This indicated that the spike was functioning properly. However, the point of confusion is whether setting "henvcfg.pbmte" under a cleared "menvcfg.pbmte" would actually take effect. This ambiguity has implications for the process of reading "henvcfg" when it is required.

rbuchner-aril, 2023-11-13:
So to clarify, when `menvcfg.pbmte` is set to 0, is should not clear `henvcfg.pbmte`. However, if `menvcfg.pbmte` is 0, then we should not be able to set `henvcfg.pbmte`.

This bug likely also effects the `STCE` and `ADUE` bits.

rbuchner-aril, 2024-01-12:
See section 2.4 of the Privileged Spec:
```
As a special case of this rule, the value written to one CSR may control whether a field of a second CSR is writable (with multiple legal values) or is read-only. When a write to the controlling CSR causes the second CSR’s field to change from previously read-only to now writable, that field immediately gets an unspecified but legal value, unless specified otherwise.
```

I believe this implies Spike's current behavior is legal (as is the behavior you expected). I do not think the above has any requirement of determinism. @viktoryou please confirm and close.

viktoryou, 2024-01-16:
Thanks a lot. The explanation is reasonable for me.

closed 2024-01-16 by viktoryou