riscv-software-src/riscv-isa-sim #1499: FPU CSRs are not logging in a standard order

omerguzelelectronicguy, 2023-11-10:
Spike dumps the "core", hart id, privilege mode, PC, instruction, register(GPR FPR CSR) if an update exists, and then memory if an update exists. The order of those things is not changed in any log as I saw. 

Sometimes two or three CSR updates are logged. For example, When we write the fcsr directly both flags and frm are updated. As I run and see in my tests the dumping order of those registers is not standard(even though the instructions are the same). Sometimes fflags are logged first sometimes the other one. Is there a specific reason for their behaviour like that or only a lack of standardization( since it is not so important)?

YenHaoChen, 2024-08-09:
Currently, the logging data structure is ```std::unordered_map```, meaning that we cannot expect the ordering for the register logging.

Regarding the GPRs, FPRs, and CSRs, their inherent order may not directly correlate with instruction execution, so displaying them in a specific order might not provide additional clarity. However, sequential updating could enhance user understanding when it comes to VRs. Displaying VRs in an orderly fashion could be beneficial. It would allow users to track the progression of vector operations more clearly.

Additionally, the current implementation uses a single data structure for logging, i.e., ```commit_log_reg_t```, with a special encoding to identify GPR, FPR, CSR, or VR. Using separate data structures for different register types instead could provide better readability on the source code.

@aswaterman How do you feel about this?

aswaterman, 2024-08-09:
In general, unordered maps should not be used for iteration, only for lookups.

Over time, it might make sense to make other improvements to the commit log, but for now, I propose we make the one-line change of using an ordered map instead of an unordered map.  This won't have any material impact on performance since the runtime is dominated by stringifying/printing the log.

PR #1768 (2024-08-09): Use ordered map for commit log
In general, unordered maps should not be used for iteration, only for lookups.

In this case, using an ordered map guarantees that the order in which writes are logged is consistent for a given instruction.

Resolves #1499
commit 8b05d84e (2024-08-09): Use ordered map for commit log

In general, unordered maps should not be used for iteration, only for
lookups.

In this case, using an ordered map guarantees that the order in which
writes are logged is consistent for a given instruction.

Resolves #1499
closed 2024-08-09 by aswaterman