riscv-software-src/riscv-isa-sim #1502: [Feature Request] Remove the limitation of VLEN <= 4K

sequencer, 2023-11-14:
We are designing [t1](https://github.com/chipsallliance/t1), which VLEN can be configured to be larger than 4K, in RISC-V Vector specification, The VLEN can reach to 64K:
https://github.com/riscv/riscv-v-spec/blob/fc76ec73fc1dd4531360ee8f3138f79f02e8b1b0/v-spec.adoc?plain=1#L67-L72
However riscv-isa-sim has this special limitation: 
https://github.com/riscv-software-src/riscv-isa-sim/blob/be5dee0bafb413c9ac8845ca144db9b7641941b2/riscv/processor.cc#L175-L177
Thus spike cannot emulate the RISC-V Vector behavior which VLEN is larger than 4K.

We wanna request spike to remove this limitation.

luojia65, 2023-11-14:
I'd like to see this requirement relaxed to at least 65,536 bytes for that we need to compare [rustsbi/zemu](gitee.com/rustsbi/zemu) to an existing RISC-V software emulation model to ensure security critical firmware implementations work as expected.

aswaterman, 2023-11-14:
Of course, I have no objection with lifting this restriction, but you will need to contribute it (after testing it thoroughly).