riscv-software-src/riscv-isa-sim #1504: RISCV32 mhpmcounter3h-mhpmcounter31h counter CSRs not working

youssefM1999, 2023-11-15:
I am trying to cause an overflow in 32 bit architecture. Writing to the lower 32 bit csr register mhpmcounter3 works; however; whenever I write to the mhpmcounter3h register, the value stays at 0. Even when the mhpmcounter3 overflows, the value of mhpmcounter3h does not change either.

Are these CSRs not implemented currently in Spike? If not, where should I add its implementation.

chihminchao, 2023-11-20:
The performance counter number and behavior are implementation-defined.  In current spike, counter number >= 3 are implemented as constant zero register (write allow read zero).  You need to modify 
https://github.com/riscv-software-src/riscv-isa-sim/blob/4841ad0238f0b71ca86fb28974765495cc0c34a9/riscv/processor.cc#L261

Abhiram1221, 2024-01-09:
> I am trying to cause an overflow in 32 bit architecture. Writing to the lower 32 bit csr register mhpmcounter3 works; however; whenever I write to the mhpmcounter3h register, the value stays at 0. [...]

Hey Youssef,
 Where can i know what mhpmevent values pair with what instructions for me to increment counter and to cause overflow based on sscofpmf extension

jason23g, 2025-02-18:
Hello @chihminchao ,

 In the current spike, counter numbers >= 3 are still being implemented as constant zero registers (write allow read zero). If so, I would like to know if I need to change a variable or if I need to put the counter name followed by 4 whenever 3 is in the counter name in the code section below.

https://github.com/riscv-software-src/riscv-isa-sim/blob/206268cfdfa55ae4c85713e201107ec9890d5756/riscv/csr_init.cc#L65-L73

Thanks in advance for your time and your effort.