riscv-software-src/riscv-isa-sim #1508: Signature memory regions

SyedHassanUlHaq, 2023-11-17:
Are there any specific memory regions used for signature dump or does it depends on the test suites and the isa? If there is a specific memory region for a signature dump then what is it?

jerryz123, 2023-11-19:
The signature region is specified by special symbols in the ELF file.

https://github.com/riscv-software-src/riscv-isa-sim/blob/4841ad0238f0b71ca86fb28974765495cc0c34a9/fesvr/htif.cc#L164C1-L169C1

SyedHassanUlHaq, 2023-11-20:
Thanks  @jerryz123, But what if there's a test case in the disassembly such as: 
```
lui	t5,0x413f3
addi t5,t5,-926
lui	a7,0xe2fe6
addi a7,a7,1091
add	zero,t5,a7
sw	zero,8(ra)
auipc ra,0x3
addi ra,ra,-564
```

here zero is being stored at address 0x00003030, the signature regions specified in the the disassembly also has this region specified but the refrence model signature file seems to have skipped this signature dump

> The signature region is specified by special symbols in the ELF file.
> 
> https://github.com/riscv-software-src/riscv-isa-sim/blob/4841ad0238f0b71ca86fb28974765495cc0c34a9/fesvr/htif.cc#L164C1-L169C1

closed 2024-09-09 by SyedHassanUlHaq